rkaczorek/astroberry-server #87: QHY 5III 462 not yet supported

MLKutzke, 2020-08-05:
I just received the new qhy camera 462 but it is not yet recognized by the astroberry.
There seems to be a new Recent driver available from qhy that supports it.

I am Using Astroberry 2.0.3 on rpi 4b.
-Mike

rkaczorek, 2020-08-05:
Thanks for reporting. I'm planning updates in a couple of days. I will check QHY driver status.

rkaczorek, 2020-08-09:
QHY drivers has been updated. Please upgrade your system and report back.

MLKutzke, 2020-08-12:
Success!
Just plugged it in, restarted the RPI4 after the update and it was recognized!
As I am typing here it is now taking the first (dark) frames (since it is cloudy it wont see first light outside tonight...).
I will report if I run into any additional issues - but this one can be closed.
-Mike

closed 2020-08-12 by MLKutzke