rkaczorek/astroberry-server #89: Astroberry unable to connect to ASI462MC

mbaddah, 2020-08-15:
I've recently purchased an ASI462MC osc ZWO camera and none of the programs seem to detect the camera at all.

I'm using a Raspberry pi4. Have installed latest Astroberry 2.0.0, run both `sudo apt update` and `sudo apt upgrade` .

e.g. If I open oacapture 1.8.0 and click on Camera, I see ZWASI2. If I click on that an error pops up saying `Unable to connect camera`.

rkaczorek, 2020-08-23:
What is the error when connecting INDI ASI driver?
oaCapture uses it's own libraries for ASI so check if camera is supported on https://github.com/openastroproject/openastro

mbaddah, 2020-08-25:
Hi @rkaczorek I just had a look at openastro code and it looks like its not supported yet. Hopefully in the next release :)

closed 2020-09-03 by mbaddah