rkaczorek/astroberry-server #91: "Service Not Available" connecting to hotspot

Abelink23, 2020-08-16:
I made a fresh installation of astroberry on my Raspberry Pi3 b+ 3-4 months ago and everything has worked perfectly but today I decided to update it via apt update/upgrade and I am no longer able to connect via browser (https://astroberry.local/desktop/ or https://http://10.42.0.1/desktop).

I get this error message:
    Service Not Available
    Requested service is not available.

    If you see this page while using floating menu, this means that system service providing this functionality is stopped.
    Check and/or restart respective system services and try again.

    Otherwise it means that your system is misconfigured. Review and/or revert your recent changes and try again.
    Review system documentation and if you believe it is a bug, report it here.

The hotspot itself works fine as I am able to connect via VNC viewer. 

Any tips about what can I do to fix it?

Many thanks!
Abel

rkaczorek, 2020-08-23:
Clear cache in your browser. Otherwise a browser can serve old content to you. You can do it by refreshing the page a few times.
If it doesn't help you need to restart respective services e.g. if GPS panel is not available run  `sudo systemctl restart gpspanel.service` This also applies to astropanel.service and indiwebmanager.service

rkaczorek, 2020-09-07:
What's the status of this issue on your side?

Abelink23, 2020-09-08:
I still have to test it and I was actually planning to do it today. I 
will update this as soon as I do it.

Abel

Abelink23, 2020-09-08:
I have tried all the commands and they all get stuck running forever. I 
am afraid that there is another program preventing their execution or 
something like that.

Also, I have tried to access via different devices with the same problem 
and in all cases clearing the cache.

Any clue of what could it be?

Many thanks,

Abel

rkaczorek, 2020-09-08:
Please take a look at this thread https://indilib.org/forum/astroberry/7627-raspberry-pi-trojan-going-wild.html

Abelink23, 2020-09-08:
I can confirm that I had many bins.deutschland-zahlung.eu entries and I 
have now changed the passwords.

What would be the next step now in order to recover the normal behavior 
of my system?

Many thanks,

Abel

rkaczorek, 2020-09-09:
The easiest way is to reflash your microSD card. The hard way is to remove the malware by hand as described here https://www.tobsan.se/update/2017/11/06/rpi-trojan.html

Abelink23, 2020-09-09:
May I ask for me to understand, what is the connection between the 
problem I experienced not being able to use the web-interface, and the 
Trojan?

Many thanks,

Abel

rkaczorek, 2020-09-09:
The trojan changes records in /etc/hosts and overtakes some localhost services. As the result web panels are not accessible on localhost (127.0.0.1)

closed 2026-02-11 by rkaczorek