rkaczorek/astroberry-server #92: Missing Symbols for indi_gpsd and indi_qhyccd.

AradoSKY, 2020-08-24:
After updating to INDI 1.8.6, both indi_gpsd and indi_qhyccd started throwing SYMBOL errors. Specifically:

2020-08-17T00:49:37: Driver indi_gpsd: indi_gpsd: symbol lookup error: indi_gpsd: undefined symbol: _ZN4INDI13DefaultDeviceD2Ev

and

2020-08-17T00:54:27: Driver indi_qhy_ccd: indi_qhy_ccd: symbol lookup error: indi_qhy_ccd: undefined symbol: _ZN4INDI7CCDChip8setFrameEiiii

A review of libindidriver.so, using nm, does not contain these symbols. A review of these drivers installation files does not have these symbols defined.

This appears to be INDI asking for data from these drivers and the Drivers cannot answer. Thank you checking.

The OS is Raspbian and the PI is an RPI4B with 4G. I have ensured all possible libraries are installed on the RPI4B.

rkaczorek, 2020-08-24:
INDI 1.8.6 has not been released for Astroberry Server yet so I don't know where you got it from.
I'm not going to support custom mods to Astroberry Server or generic linux, INDI issues. It is up to a user to handle such issues.

closed 2020-08-24 by rkaczorek

AradoSKY, 2020-08-24:
Hello Radek

Hmm... all I did was “apt-get update && apt-get upgrade”. This occurred after 1.8.6 was released August 12. How may I stop getting INDI updates to prevent my system getting unsupported INDI updates?

AradoSKY, 2020-08-24:
root@astroberry:/home/astroberry/AstroRoot/indi-3rdparty/indi-gpsd# apt-get update && apt-get upgrade
Hit:1 http://archive.raspberrypi.org/debian buster InRelease                                                                                       
Hit:2 https://www.astroberry.io/repo buster InRelease                                                                                              
Hit:3 http://raspbian.raspberrypi.org/raspbian buster InRelease                                             
Hit:4 http://vega.ap-i.net/apt unstable InRelease  

Hello Radek

Which one of these allowed 1.8.6 to be installed? I received this the same day 1.8.6 was released. I do not have a custom configuration. Somehow, I received a work in progress or a path reaches the INDI release bin.

rkaczorek, 2020-08-24:
I believe none of the above serves INDI packages except https://www.astroberry.io/repo
You must have added some other repo e.g. ppa:mutlaqja/ppa or stellarmate. You need to review /etc/apt/sources.list and /etc/apt/sources.list.d/
I will be releasing INDI 1.8.6 soon. All drivers must be released along new INDI version so that is why you see missing symbols after upgrade from 1.8.5

AradoSKY, 2020-08-24:
Hello Radek,

I have reviewed both files, sources.lst and Astroberry list. This is its content.

deb https://www.astroberry.io/repo/ buster main

I have just three repositories to pull from. The other two are Raspbian repositories. This where my AstroBerry is at the present. Once updated, backing out is not possible.

AradoSKY, 2020-08-26:
Hello Radek,

No. I do not have a radical setup. I don’t have PPA setup. In fact, the Raspberry PI fo AstroBerry installs 1.8.6. Reinstalling libindi1 installs 1.8.5. Looks like a version control issue. I am looking forward to a working 1.8.6. Thank you for your support. I have no way past beside flattening and rebuilding.