rkaczorek/astroberry-server #94: Not booting after update & upgrade

M31Fan, 2020-08-25:
Hello, 

I couldn't boot to my system after a rootine update / upgrade. I thought I had made a mistake some where, so I reflashed my card with the Astroberry I already had on my laptop (2.0.1). It stopped working again immediately after upgrade and upload. I tried it a couple more times, and also with the latest 2.0.2 version. All worked well until the update+upgrade, after which it didn't boot. 
I tried to check if it was with the hotspot, but nothing. The Raspberry flashed green 4 times, which indicates some sort of boot error (start.elf not launched, according to https://elinux.org/R-Pi_Troubleshooting#Green_LED_blinks_in_a_specific_pattern). 

I have no idea of what's the cause of error, but I'm fairly confident something went wrong on the respository...

Did anyone else encountered the same problem?

Many Thanks!
Tiago

rkaczorek, 2020-08-26:
Have you connected to a display via HDMI to observe booting process? Please provide pictures of the screen.

M31Fan, 2020-08-26:
Unfortunately I didn't have enough time, but I'll try as soon as I can.

cplonski, 2020-08-26:
[image]
Here is what I'm seeing

rkaczorek, 2020-08-26:
Thanks for reporting this issue. This seems to be related to the latest upgrade of raspbian kernel by Raspberry Foundation. Looks like upgrade from the previous version (used in Astroberry 2.0.2) fails for some reason. I need to investigate this and reproduce the issue on my side.

cplonski, 2020-08-26:
Just FYI, I got a generic 32GB microSD and it boots up just fine!

M31Fan, 2020-08-26:
Even after update and upgrade? 
If yes, it could be related to the type of partition.

cplonski, 2020-08-26:
yes, same exact image (2.0.2) works on 32GB but not on 64GB

rkaczorek, 2020-08-26:
@M31Fan can you confirm it with 32GB microSD card?

M31Fan, 2020-08-26:
I’m not sure if I have one at home. 
I can only find 64 and 128 cards... I’ll try!

M31Fan, 2020-08-27:
I think it's solved.
I tried on a 16Gb card worked well. I did however made something different.
So I repeated the process on a 128 card, and it now works well too.

The aspect is that during the upgrade process it pauses around 63% and asks
about the hotspot connections, and in my previous attempts I had selected
option D to see the differences. I then tried to keep the existing wifi
settings and move on with the process, but somehow had failed.
On my 16 Gb Card I selected the option N (no changes to the wifi settings)
and the process concluded well and was able to reboot without any issues.
I then reflashed again my 128 Gb card and went through the process. Now I
have the astroberry 2.0.3 working.

The key aspect is to stick with the default answer when it asks about the
hotspot  connection.

Many Thanks!

closed 2020-09-07 by rkaczorek