rkaczorek/astroberry-server #97: GPS Panel not fixing to virtualgps

festoon1, 2020-09-01:
When I reboot my pi running astroberry server v2.0.3 - if I look at the GPS panel it says Fix Waiting. Now if I connect through the terminal and restart the virtualgps, then it finds it gps fix correctly as I inputted from the Preferences/Geographic Location menu. All at this point is fine. The problem is that every time I reboot I then find the gps panel goes back to says Fix Waiting.
Any ideas how to resolve the issue?

rkaczorek, 2020-09-02:
Is virtualgps enabled at boot? You can check it after starting your system with `sudo systemctl status virtualgps`
If it is not, you can enable it with `sudo systemctl enable virtualgps`

festoon1, 2020-09-02:
Thanks...when I run sudo systemctl status virtualgps from terminal it tells me the service is active running...which is the strange thing. However the GPS panel just flashes with waiting
[image]
[image]

festoon1, 2020-09-02:
Also just in case I've tried sudo apt update and sudo apt upgrage to be sure this was not to issue, but it did not solve the problem

rkaczorek, 2020-09-02:
Please check the status of gpsd.service as well and finally verify if running `gpsmon` from terminal gets the fix

festoon1, 2020-09-02:
Thank you again. 
The gpsd status gives
[image]
However running gpsmon from terminal does not get a fix
[image]

rkaczorek, 2020-09-02:
Thanks for additional data.
The file /tmp/vgps is should be created by virtualgps.service - it is the source of gps data stream for gpsd service. Apparently it is not there. You need to inspect this further:

- Stop virtualgps service: `sudo systemctl stop virtualgps`
- Check if the file exists: `ls -al /tmp/vgps`
- If it exists, remove it: `rm -rf /tmp/vgps`
- Start virtualgps service: `sudo systemctl start virtualgps`
- Check if the file exists, also note file permissions: `ls -al /tmp/vgps`

If it does not exist or file permissions does not allow for reading the file by gpsd user, there is some kind of issue with virtualgps.service itself. Otherwise check gpsmon again and this time it should work

festoon1, 2020-09-02:
When I stop the virtualgps and check if the file /tmp/vgps exists it is not there. Then if I startvirtualgps and check again /tmp/vgps is there
[image]
However upon reboot gps still does not find a fix, but file now does exist
[image]

rkaczorek, 2020-09-02:
So it seems OK. Does gpsmon work now?

festoon1, 2020-09-02:
Unfortunately not - when I restart the gps does not fix - as in the last picture. Basically I cannot get gps to fix whenever I restart the pi. (unless I go through and restart virtual gps). Any suggestions?

rkaczorek, 2020-09-07:
Try stoping virtualgps service and then run it manually `/usr/bin/python3 /usr/bin/virtualgps.py`. Please report any suspicious output
What's you /etc/default/gpsd content?

capape, 2021-01-05:
Hi,

I had the same problem.

I have this output
ene 05 19:42:53 astroberry systemd[1]: Started GPS Panel.
ene 05 19:43:00 astroberry python3[1570]: The client is using an unsupported version of the Socket.IO or Engine.IO protocols (further occurrences of this error will be logged with level INFO)

rkaczorek, 2021-02-22:
Any update on this issue?

prdgymx, 2022-03-14:
Just a heads up for anyone else, I was able to get the GPS map to properly point to my location after clearing site cookies. Assuming astroberry.local is storing location info there?

Either way, once I cleared cookies (or loaded in private browsing tab) the map showed my proper location.