rkaczorek/astroberry-server #100: FireCapture v2.6 in Astroberry 2.0.3 does not work on ZWO ASI462MC

lucian-liu, 2020-09-23:
I know that this is a quite known issue by many FireCapture v2.6 users ==> ZWO ASI462MC can not be recognized by it!!!
But I just want to know if there is any update plan for Astroberry server to upgrade FireCapture to V2.7 beta or later. This would be great if I can know when the newer version would be available.
FYR that I also updated my astroberry with apt update and upgrade right after I flashed the 1st image.

elenin, 2020-10-12:
Also not work on QHY 462C,tested on astroberry with raspi and ubuntu20.04 with intel i7.

rkaczorek, 2020-10-12:
Firecapture uses own drivers. Please report unsupported devices to http://www.firecapture.de/

TonyF1330, 2022-08-28:
FireCapture 2.7 (release, not beta) has been available as of March 2022 and appears to have resolved the issue with ZWO cameras. Can version 2.7 be bundled/added to the next Astroberry release instead of 2.6.x?

I have no idea how difficult this ask is so I apologize if this is a huge milestone that takes lots of time to plan and implement.

Thanks!

lucian-liu, 2022-08-29:
2.7 beta has fixed this issue but it still requires manual installation in Astroberry. However, I am good to close this issue.

closed 2022-08-29 by lucian-liu

TonyF1330, 2022-08-29:
I'm not very knowledgeable regarding manual installations on Astroberry (Raspberry Pi or Linux for that matter). Is there a "dummies" guide that I can follow "step by step" to manually install FC 2.7? I've tried downloading 2.7 from the FC website and installing it. It appears to finish installing but then won't load when I try to run it. I'm sure that I'm missing a step or doing something wrong. Thanks!

rkaczorek, 2022-08-29:
Firecapture 2.7.10 has bee just pushed to astroberry repository. Run `sudo apt update && sudo apt upgrade` to get it

lucian-liu, 2022-08-29:
This should be the steps as I recalled.
1. Download FireCapture for RPi ([link](https://drive.google.com/file/d/1qD7v9aW-y6yg_UA347JXBoNVNKNkt1sR/view?usp=sharing))
2. Open "Terminal" to install FC v2.7 by
> sudo su
> dpkg -i firecapture_2.7.10_armhf.deb
3. After you install FC, you should also modify its shortcut.
> Find FC shortcut from Astroberry start menu.
> Right-click FC shortcut to access "Properties"
> Change "Command" and "Working directory" to where your FC is.
Then you should be good to run FC v2.7.

lucian-liu, 2022-08-29:
[image]
I have the snapshot of where to set FC v2.7 shortcut here.

TonyF1330, 2022-08-29:
After the update to 2.7 using `sudo apt update && sudo apt upgrade` FireCapture 2.7 doesn't launch successfully for me. If I turn on the start menu option to run in a terminal window I get the following output. This is similar to the errors that I was seeing when I tried to manually update before asking for help. I wonder if I messed something up during those earlier attempts to get 2.7 to work? I'm considering installing a fresh image unless anyone has any suggestions that I should try first.

2022-08-29 13:57:31.589 INFO de.wonderplanets.firecapture.log.Log logInfo Starting FireCapture - lets go....
2022-08-29 13:57:31.676 INFO de.wonderplanets.firecapture.log.Log logInfo System information
  FireCapture Release: v2.7 
  FireCapture Build: 10 
  FireCapture Directory: /opt/FireCapture_v2.7 
  OS: Linux
  OS Version: 5.10.103-v7l+
  PATH: /usr/java/packages/lib:/lib:/usr/lib
  JRE Version: 12.0.2-BellSoft
  JRE ClassPath: lib/radiance-substance-3.0.0.jar:lib/radiance-neon-3.0.0.jar:lib/radiance-trident-3.0.0.jar:lib/swingx-1.6.jar:lib/zstd-jni-1.3.3-4.jar:lib/loci-common.jar:lib/Firecapture.jar:lib/fits.jar:lib/ome-xml.jar:lib/ij.jar:lib/Filters.jar:lib/INDI.jar:lib/jcalendar-1.3.3.jar:lib/RXTXcomm.jar:lib/bio-formats.jar:lib/jgoodies-common-1.3.0.jar:lib/jSerialComm-1.3.11.jar:lib/jacob.jar:lib/jgoodies-forms-1.5.0.jar:
  JRE initial heap size: -Xms1024m
  JRE max heap size: -Xmx1024m
  Camera: 0

2022-08-29 13:57:31.678 INFO de.wonderplanets.firecapture.log.Log logInfo Loaded settings from: /home/astroberry/FireCapture/FireCapture.ini
2022-08-29 13:57:31.681 INFO de.wonderplanets.firecapture.log.Log logInfo Loaded FireCapture.ini
java.lang.ClassNotFoundException: org.pushingpixels.substance.api.skin.SubstanceOfficeSilver2007LookAndFeel
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:583)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:332)
	at de.wonderplanets.firecapture.util.GuiUtil.setLookAndFeel(GuiUtil.java:193)
	at de.wonderplanets.firecapture.profile.SystemSettings.initProfile(SystemSettings.java:906)
	at de.wonderplanets.firecapture.profile.SystemSettings.getCurrent(SystemSettings.java:1577)
	at de.wonderplanets.firecapture.capture.Capture.<clinit>(Capture.java:60)
	at de.wonderplanets.firecapture.profile.SystemSettings.<clinit>(SystemSettings.java:116)
	at de.wonderplanets.firecapture.gui.FireCapture.<clinit>(FireCapture.java:165)
Exception in thread "main" java.lang.ExceptionInInitializerError
	at de.wonderplanets.firecapture.profile.SystemSettings.<clinit>(SystemSettings.java:116)
	at de.wonderplanets.firecapture.gui.FireCapture.<clinit>(FireCapture.java:165)
Caused by: java.lang.NullPointerException
	at de.wonderplanets.firecapture.profile.SystemSettings.initCaptureProfiles(SystemSettings.java:1367)
	at de.wonderplanets.firecapture.profile.SystemSettings.initProfile(SystemSettings.java:912)
	at de.wonderplanets.firecapture.profile.SystemSettings.getCurrent(SystemSettings.java:1577)
	at de.wonderplanets.firecapture.capture.Capture.<clinit>(Capture.java:60)
	... 2 more

rkaczorek, 2022-08-29:
I would suggest removing all settings by running `rm -rf /home/astroberry/FireCapture`
Then, try it running with your camera disconnected. It should start with demo mode.
If it is ok rerun with a camera. Whatever happens provide us with feedback.

PS. What's your hardware? It is always worth mentioning as it helps diagnosing an isse.

TonyF1330, 2022-08-29:
Perfect, that procedure solved my issue. Thank you again!
For reference, I have an ASI585MC.