rkaczorek/astroberry-server #102: Unable to connect PS3 Bluetooth controller to Astroberry

michulus, 2020-10-02:
Hi
Sony PS3 controller in pairing mode is not detected by Astroberry bluetooth (add device). The same controller works correctly when connected via USB. I tried already with a few PS3 controllers, non of them is BT discovered by Astroberry.
If this is configuration issue please share the solution for it.
Thank you

rkaczorek, 2020-10-02:
Have you tried the following guide? https://pimylifeup.com/raspberry-pi-playstation-controllers/

michulus, 2020-10-02:
Yes, I did. In my case [sixpair] displays MAC for BT RPI controller instead of PS3 MAC, which is trying to pair.
Bluetooth configuration tool [bluetoothctl] scan shows all devices in the area but not PS3 in pairing mode.
I also tried to install [SIXAD] but the procedure ended with BT RPI controller unable to detect any device in the area.

rkaczorek, 2020-10-02:
I'm afraid I cannot help with it. It is controller-to-os issue. It must work with operating system before it can be used by indi driver

michulus, 2020-10-03:
Radek,

After extensive research and multiple try and error attempts, I was able to successfully pair PS3 to Astroberry using BT procedure described in the following URL:  https://www.piborg.org/blog/rpi-ps3-help

Please note though, that above procedure has some missing steps, which are crucial for success of the whole operation.
In my case, after successful execution of [bluetoothctl] commands to [connect] and [trust] PS3 device ), I had to:

1. Open BT manager menu and click ADD device. 
2. Select MAC address of PS3 controller.
3. When BT manager asked for PIN, quickly connect PS3 to USB and allow pairing process to complete.

Sharing this in case if somebody need's it 

BR
Chris