rkaczorek/astroberry-server #103: Wifi to home network creates a new MAC address every boot - so new IP address as well

pootle, 2020-10-05:
I put a wpa_supplicant file (with my wifi details) in the boot partition as I normally do for a new SD card with Rspberry Pi OS.

Now every time I boot I get a new IP address (and it's a new MAC address), and the wifi icon top right shows no network connections at all.

rkaczorek, 2020-10-05:
Astroberry uses NetworkManager instead of bear bone wpa supplicant. You should not use wpa supplicant configuration file. Instead configure your network setting with GUI or by files located in /etc/NetworkManager/system-connections/

pootle, 2020-10-06:
Ah! re-installed astroberry without suppllicant file and after a bit of fiddling with network manager, all is well. Thanks

closed 2020-10-06 by pootle