rkaczorek/astroberry-server #104: Speedport wifi connection

rkaczorek, 2020-10-06:
I work with an raspi pi 4 and astroberry. Up to now I am very satisfied. But two problems are left:
1. I cannot connect the raspi with my router speedport of german telecom.
2. I have an spectrograph on my telescope. Between telescope and spectrograph is a motor driven flip mirror which is controlled by an arduino. The arduino is with usb connected to the raspi. After 10 frames taken by the scientific camera I want to take a calibration frame the flip mirror must be flipped for. How can I program the observation planer?

_Originally posted by @Lezten in https://github.com/rkaczorek/astroberry-server/issues/74#issuecomment-704348123_

rkaczorek, 2020-10-06:
Re #1 - How do you try connecting to Speedport router?
Re #2 - How do you control arduino based flip mirror when connected to a PC? Normally you need a piece of software on PC side to control arduino or you can send a command via serial connection. To control it from INDI you would need an INDI driver which probably does not exist. It can be probably easy to develop, depending how you control it from PC.

Lezten, 2020-10-06:
Thank you for your quick response.
1. I use the WLAN controler where I can see all available networks. I choose my Speedport and press the conection body. After some seconds appears a message "connection not possible". I have the same problem when I try it from my notebook under windows. My notebook is a dual boot system. When I choose ubuntu 20.04 I have no problem to connect with the raspi. As long as the raspi is working I have a problem to connect to my speedport too under ubuntu. When the raspi is off there is no problem.
2. The flip mirror is driven by a stepper (CheapStepper) from seeed. The stepper is connected to a stepper driver which im turn is connected to an arduino uno. The arduino receives the signals "1" (= up) or "2" (=) down via the the serial board of the arduini IDE.

rkaczorek, 2020-10-09:
You can try manually configuring a connection to the router. Instead scanning available networks, right-click wlan icon and edit connections. Change select a wireless connection and edit SSID and password parameters.
Regarding flip mirror, you need to control it the same way. The fast test would be using `miniterm /dev/ttyUSBx` or whatever your serial device is for arduino connected via USB. Then you can write values over serial as in Arduino IDE. The better way is to write a script which will do it for you. The best way would be to create an INDI driver that will do it for you.

jscheidtmann, 2021-02-14:
Check your network connection and IP settings. Are you using fixed IP addresses?