rkaczorek/astroberry-server #109: ASI120MC USB2.0 and the newer Linux kernel

jojje70, 2020-11-15:
After I upgraded the sd memory card for my RPI3 to the latest image of astroberry 2.0.3, my ASI120MC USB2.0 (default firmware) stopped working completely (worked neither as a regular camera nor a guide camera). Tried to flash the camera to the compatible version (ASI120MC-compatible.iic) that ZWO provides and got the camera running again, but random timeouts and split frames were introduced in both Ekos and PHD2 which I did not have before. Then I tried to revert the Linux kernel to version 4.19.118 and re-flashed the camera back to the default firmware. After that my camera performed as usual again. 

Something is problematic with the latest kernel in combination with the older ASI120MC USB 2.0. Maybe it´s someting that is not fixable by driver as it appears tom be a problem with the newer Linux kernel in combination with this particular camera?

/Jojje

rkaczorek, 2020-12-06:
Yes, it is kernel related. USB 2.0 ASI is somehow unstable. USB 3.0 works without issues

jojje70, 2020-12-09:
I did a quick test on RPi4 where I get timeouts/split frames intermittently no matter which Linux kernel. RPi4 seems to handle USB differently than RPi3, where it actually works with default ASI120 firmware and an older Linux kernel. The solution seems to be to avoid ZWO ASI USB2 cameras in the future. I will close this case.

closed 2020-12-09 by jojje70

chase996, 2021-02-28:
Is there a command you entered on the pi to revert the kernel back?

jojje70, 2021-03-02:
> Is there a command you entered on the pi to revert the kernel back?

"sudo rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2" to roll back to 4.19.118

chase996, 2021-03-02:
> > Is there a command you entered on the pi to revert the kernel back?
> 
> "sudo rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2" to roll back to 4.19.118

Super helpful. I was able to get my 120mm up and running with my pi. 
Thank you!