rkaczorek/astroberry-server #110: Phd2 does not start on raspberry zero W

HaraldStauss, 2020-11-18:
Using version 2.0.3 on a raspberry zero W, PHD2 does not start. I wonder if it has to do with the ARM6 processor vs. the ARM7 processor used in other raspberry devices.

rkaczorek, 2020-12-06:
Can you provide some details? Try running phd2 from terminal and provide the output

HaraldStauss, 2020-12-07:
The error message is "Invalid Instruction". 
After discussing this with Patrick, it seems like the problem is that the QHY SDK is compiled for armv7 and the raspberry zero W has an armv6 processor. Thus, the QHY library is incompatible with the PI Zero. I have contacted QHY and they are now looking into compiling their SDK for the armv6. Once this SDK is available, the problem can be fixed, I think.
Thanks for your reply and for making astroberrry available.
Harald

HaraldStauss, 2020-12-09:
I can now confirm that the armv6 QHY library that QHY sent me works on the Raspberry PI Zero. I have not tested it on Astroberry yet, but I was able to access my QHY 5-III 462C through the INDI driver in KStars/EKOS on a standard (not Astroberry) system on my Raspberry PI Zero W. 
To test it on Astroberry, would it be sufficient to replace the library? In which directory is it installed on Astroberry?
Thanks,

Harald

rkaczorek, 2020-12-09:
AFAIK PHD2 does not use indi libs and provides its own qhy library. Take a look PHD2 project [here](https://github.com/OpenPHDGuiding/phd2/tree/master/cameras/qhyccdlibs)

HaraldStauss, 2020-12-09:
I was under the assumption that both the INDI driver and the native driver of PHD2 rely on the libraries provided by the QHY SDK.

rkaczorek, 2020-12-09:
They do, but INDI provides them with libqhy package and PHD2 includes them at compilation time, not using libraries provided by libindi

HaraldStauss, 2020-12-10:
PHD2 has a native driver for QHY that relies on the QHY SDK but it also allows access to QHY cameras through the INDI driver (which also relies on the QHY SDK). Since the raspberry PI Zero has an armv6 processor and not the armv7, the QHY SDK library for armv6 is required either way.

rkaczorek, 2020-12-10:
This all true. Your question was "Using version 2.0.3 on a raspberry zero W, PHD2 does not start". The fact that PHD2 can use a camera via INDI driver has nothing to do with the issue reported, does it?
Astroberry provides official INDI drivers and PHD2. If they support armv6 from QHY SDK, respective updates will be available for Astroberry

HaraldStauss, 2020-12-10:
Hi:
First of all, I would like to say that astroberry is absolutely great!!!
The issue is that some software bundled with astroberry utilizes libraries that are incompatible with the armv6 processor on which the Raspberry PI Zero/Zero W is based upon. As a result several software, including PHD2 that are included with astroberry do not run on a Raspberry PI Zero. To make it work on the Raspberry PI Zero, these libraries would have to be replaced with armv6 compliant libraries. But ....

I don't think this would be a good solution, because it will likely mean that some of the software will not run as efficient on Raspberry PIs that are based on the armv7 processor. Quite frankly, I think the Raspberry PI Zero is not powerful enough to run astroberry. Maybe instead of downgrading the libraries to armv6 compliant ones, the best solution would be to include a disclaimer in the documentation of astroberry to indicate that astroberry is not fully compatible with the Raspberry PI Zero because of the armv6 vs. armv7 processor.

Once more thank you for putting together astroberry. It is a great product !!!
I have now ordered a Raspberry PI 4 to replace the Zero so that I can use astroberry!!!
Regards,
Harald