rkaczorek/astroberry-server #111: Bumping dependencies versions

tvaucher, 2020-11-18:
Hello,

First of all, thank you very much for the awesome RPi image. This issue concerns upgrading the dependencies in the image. The context is, in FujiFilm astro community, we can only rely on `libgphoto2` and the indi driver that comes with it `indi-gphoto` to automatize our sequences. However, essential functionalities were added in `libgphoto2==2.5.26` whereas, IIRC, `astroberry==2.0.3` comes with `libgphoto2==2.5.24`.

I rebuilt the dependencies from source and could make a write up for less tech savvy people, but I was wondering if it was not instead possible to upgrade them directly in the image. I am sure there are a lot of other libraries that received updates recently that would make sense to upgrade for the community.

Thanks in advance for your time and consideration.

rkaczorek, 2020-12-06:
libgphoto will be updated along other packages soon

jscheidtmann, 2021-02-15:
lingphoto2 is now 2.5.26.