rkaczorek/astroberry-server #112: Astroberry Server - advanced install - not working yet?

CaptainRon47, 2020-12-05:
12/5/2020
Trying to update to the latest Astroberry Server with Kstars 3.5.0 on an Rpi4 with 4 Gb ram.  I first tried loading Ubuntu/mate using the 'advanced' procedure (loading the OS first and upgrading).  But that failed (same as below).  Then I went back to Raspbian Buster OS and followed the same procedure and that failed as listed below:

entering:
wget -O - www.astroberry.io/repo/key | sudo apt-key add -
sudo su -c "echo 'deb www.astroberry.io/repo/ buster main' > /etc/apt/sources.list.d/astroberry.list"
sudo apt update
sudo apt upgrade

Then I typed :
sudo apt install astroberry-server-full
and it comes back and says there are unmet dependencies:
$ sudo apt install astroberry-server-full
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
astroberry-server-full : Depends: astroberry-server-wui but it is not going to be installed
Depends: indi-full but it is not installable
Depends: kstars-bleeding but it is not installable
Depends: gsc but it is not installable
Depends: phd2 but it is not installable
Depends: phdlogview but it is not installable
Depends: oacapture but it is not installable
Depends: indi-astroberry-amh but it is not installable
Depends: indi-astroberry-diy but it is not installable
Depends: indi-astroberry-piface but it is not installable
Depends: indiwebmanagerapp but it is not installable
E: Unable to correct problems, you have held broken packages.

So this procedure does not seem to be working for getting to Astroberry/Kstars 3.5.0

I was able to install Kstars 3.5.0, but not the Astorberry server using this method.
I will try burning the latest astroberry image next.

Ron

rkaczorek, 2020-12-06:
The issue here is: astroberry-server-full : Depends: astroberry-server-wui but it is not going to be installed
This is very strange as astroberry repo provides astroberry-server-wui package.
Try installing astroberry-server-wui first and then astroberry-server-full

CaptainRon47, 2020-12-06:
I believe I tried that and it would not load the missing packages independently.    I first loaded: Ubuntu-mate-20.04.1-desktop-arm64+rasp.img Note: that it is using arm64, I am wondering if this is the issue.
When I then switched to a new install of the latest Raspbian it also listed using arm64 and Astroberry would not install on their either of those systems.  On the Ubuntu/Mate OS I was able to use the ppa to install Kstars-bleeding, which is Kstars 3.5.0, but not on the Raspbian with the arm64 (it did not find the ppa, as I was told that would not work).  There is some work around, but I did not want to mess with that.

The other SD cards that I do have Astroberry Server running on are using armv71 OS. I am not well enough versed in the nuances of the interaction of the OS and programs to know if this is the real issue, but it is suggestive that it is to me. If this related to my problem, then perhaps it would be good to somehow qualify what OS version the Astroberry/Kstars suite can be installed on.  When I went back to an arm71 OS on the Rpi 4, I was able to load this version of the Astroberry Server, however Kstars was still 3.4.3.  So that was not what I wanted anyway, so I may just wait until Kstars 3.5.0 is released for the Rpi.  I am told I can build it from the source, however I cannot get this page to load.
httpss://techbase.kde.org/Projects/Edu/KStars/Building_KStars_KF5

My purpose of doing this was to get Kstars 3.5.0 (with the Stella Solver) running on the Rpi 4.  I have it running on a Win 10 x64 laptop and that works fine.  As I mentioned on the Indi Forum, it is not clear that I need the Astroberry server as I run a wired connection to the Rpi at the scope from a laptop in the house.  I can just run the Indi device manager from the command line and then connect and run Kstars over VNC.  Of course, VNC is problematic on the Ubuntu/Mate OS, so at this point I may just continue using Kstars 3.5.0 on the laptop until there is a 'stable' release of 3.5.0.

thanks,
Ron

rkaczorek, 2020-12-06:
Astroberry packages are for 32bit rasbian only. Ubuntu is not supported, the same for arm64.

CaptainRon47, 2020-12-06:
thanks,  That is pretty much what I have concluded.  The last question is can I load Kstars 3.5.0 (Bleeding) on a Raspbian 64 bit system on the Rpi?  I think eventually that is were I am headed, as I don't care for Ubuntu.

CaptainRon47, 2020-12-06:
I guess one more question, If I have Astroberry 2.0.3 running on a 32 bit Rpi with Kstars 3.4.3, how can I get Kstars (only) 3.5.0 running?   I tried update, upgrade approach but that still left me with 3.4.3 (stable).  Can I use the approach here to update to 3.5.0? 
https://www.indilib.org/download/raspberry-pi.html
And will this then work with the already installed Astroberry but overwriting the old Kstars?
Again my purpose is just to get Kstars 3.5.0 running on the Rpi (probably don't care if it is 32 or 64 bit), but I am not ready to build from the source, but I can follow the instructions from the website above.
thanks,

Ron
thanks,

rkaczorek, 2020-12-07:
You can get it now. Updates were released yesterday so run sudo apt update && sudo run upgrade