rkaczorek/astroberry-server #113: Astroberry crashing with ZWO Asi2600 MC Pro camera

Lucanerli, 2020-12-16:
Hi,
thanks for the great work on INDI
I am an amateur astronomer I would like to use Raspberry Pi3b with Kstars-Ekos on my setup but I find a problem of which I don't understand the origin. When I take a photo with my ZWO Asi 2600 MC Pro the system locks on the fits viewer. Are you aware of the problem? Could it depend on the driver, or could it be due to the low power of the RPI3?
regards Luca

rkaczorek, 2020-12-17:
I think it can be caused by limited RAM on RPI3. Disable debayering and/or enable limited resources mode in KStars setting.

Lucanerli, 2020-12-17:
Thanks
Nerli Luca
lunerli54@gmail.com
GAMP Gruppo Atrofici Montagna Pistoiese
3357286011

bekro, 2020-12-19:
The camera works perfect on the RPi4 (8GB version).

closed 2021-02-22 by rkaczorek