rkaczorek/astroberry-server #115: Astroberry not connecting USB device for GoToNova 8400

mijahauan, 2020-12-26:
I’d hoped astroberry on an RPi v4 would work relatively easily with two items:  the RPi HQ camera and my old 8400 series iOptron GoToNova telescope mount.  

I booted with the astroberry image, setup wifi, did an apt update, apt upgrade, and apt dist-upgrade, and installed the indi_rpicam and indi_lx200gotonova drivers before plugging in the iOptron and attempting to start and run INDI.  

However, I’m failing… I suspect it has to do with properly attaching the USB device but I’m no expert in this at all!

The Silicon Labs CP210x USB to UART serial driver loads and mounts(?) a device at /dev/ttyUSB0 when plugged in:

Dec 26 01:34:26 astroberry kernel: [ 4849.555130] usb 1-1.2: new full-speed USB device number 5 using xhci_hcd
Dec 26 01:34:26 astroberry kernel: [ 4849.690646] usb 1-1.2: New USB device found, idVendor=10c4, idProduct=ea6
0, bcdDevice= 1.00
Dec 26 01:34:26 astroberry kernel: [ 4849.690672] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNu
mber=3
Dec 26 01:34:26 astroberry kernel: [ 4849.690687] usb 1-1.2: Product: CP2102 USB to UART Bridge Controller
Dec 26 01:34:26 astroberry kernel: [ 4849.690701] usb 1-1.2: Manufacturer: Silicon Labs
Dec 26 01:34:26 astroberry kernel: [ 4849.690715] usb 1-1.2: SerialNumber: 0001
Dec 26 01:34:26 astroberry kernel: [ 4849.698343] cp210x 1-1.2:1.0: cp210x converter detected
Dec 26 01:34:26 astroberry kernel: [ 4849.708465] usb 1-1.2: cp210x converter now attached to ttyUSB0
Dec 26 01:34:26 astroberry mtp-probe: checking bus 1, device 5: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2"
Dec 26 01:34:26 astroberry mtp-probe: bus: 1, device: 5 was not an MTP device
Dec 26 01:34:27 astroberry mtp-probe: checking bus 1, device 5: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2"
Dec 26 01:34:27 astroberry mtp-probe: bus: 1, device: 5 was not an MTP device
Dec 26 01:34:27 astroberry systemd[1]: Starting Manage ttyUSB0 for GPS daemon...
Dec 26 01:34:27 astroberry gpsdctl: gpsd_control(action=add, arg=/dev/ttyUSB0)
Dec 26 01:34:27 astroberry gpsdctl: reached a running gpsd

Not sure why an MTP-probe happens, like it’s assuming its a storage device but it does seem to load gpsd… 

When I remove the USB cable, it disconnects and stops the GPS daemon:

Dec 26 11:13:05 astroberry kernel: [39568.291502] cp210x ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
Dec 26 11:13:05 astroberry kernel: [39568.292142] cp210x ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
Dec 26 11:13:05 astroberry kernel: [39568.520098] usb 1-1.2: USB disconnect, device number 5
Dec 26 11:13:05 astroberry kernel: [39568.520277] pps pps0: removed
Dec 26 11:13:05 astroberry kernel: [39568.520440] cp210x ttyUSB0: failed set request 0x7 status: -19
Dec 26 11:13:05 astroberry kernel: [39568.520453] cp210x ttyUSB0: failed set request 0x12 status: -19
Dec 26 11:13:05 astroberry kernel: [39568.520461] cp210x ttyUSB0: failed set request 0x0 status: -19
Dec 26 11:13:05 astroberry kernel: [39568.521950] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
Dec 26 11:13:05 astroberry kernel: [39568.522003] cp210x 1-1.2:1.0: device disconnected
Dec 26 11:13:05 astroberry systemd[1]: Stopping Manage ttyUSB0 for GPS daemon...
Dec 26 11:13:05 astroberry gpsdctl: gpsd_control(action=remove, arg=/dev/ttyUSB0)
Dec 26 11:13:05 astroberry gpsdctl: reached a running gpsd
Dec 26 11:13:05 astroberry systemd[1]: gpsdctl@ttyUSB0.service: Succeeded.
Dec 26 11:13:05 astroberry systemd[1]: Stopped Manage ttyUSB0 for GPS daemon.

However,  attempts to connect via EKOS always timeout.  

INFO    3191.344047 sec : Session log file /home/astroberry/.indi/logs/2020-12-26/indi_lx200gotonova/indi_lx200gotonova_06:28:57.log
DEBUG   3201.485369 sec : Toggle Debug Level -- Scope Verbose
DEBUG   3202.424057 sec : Toggle Logging Level -- Scope Verbose
INFO    3205.593076 sec : Connecting to 192.168.1.129@7624 ...
DEBUG   3205.593592 sec : Connection successful, attempting handshake...
DEBUG   3205.593672 sec : Initializing IOptron using :V# CMD...
ERROR   3208.595224 sec : Timeout error
ERROR   3211.648641 sec : Timeout error
DEBUG   3211.698898 sec : Handshake failed.

Attempts to connect locally, running KStars and EKOS also fail.

So, I’m at a loss at this point and hope you might provide an easy pointer to a solution.

BTW, if I plug the GoToNova into my macbook and run KStars/EKOS it loads the driver and device and I can control the mount as expected.  So, the Mac properly establishes the USB connection but Astroberry on an RPi v4 does not.  

On the RPi, lsusb -v reports:

Bus 001 Device 008: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP2102/CP2109 UART Bridge Controller [CP210x family]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x10c4 Cygnal Integrated Products, Inc.
  idProduct          0xea60 CP2102/CP2109 UART Bridge Controller [CP210x family]
  bcdDevice            1.00
  iManufacturer           1 Silicon Labs
  iProduct                2 CP2102 USB to UART Bridge Controller
  iSerial                 3 0001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              2 CP2102 USB to UART Bridge Controller
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

I hope it's just some silly mistake I'm making that someone can point out for me to fix...

Thanks!

mjh

Michael Hauan

rkaczorek, 2021-02-22:
Any update on this issue?

mijahauan, 2021-02-22:
I have not tried to fix the astroberry distro so it loads the lx8400 driver properly.

What I found worked was to use a regular raspberian distro and load the astroberry apt info and install only the components I needed for my need.  The LX8400 driver is in the Indi-full package and without the other stuff astroberry adds in its distro, it loads fine.  I’m using the new rpicam so I load that package, too.

Cheers!

mjh

wget -O - https://www.astroberry.io/repo/key | sudo apt-key add -
sudo su -c "echo 'deb https://www.astroberry.io/repo/ buster main' > /etc/apt/sources.list.d/astroberry.list"
sudo apt update
sudo apt upgrade
sudo apt install indi-full
sudo apt install indi-rpicam
sudo apt install indiwebmanagerapp

closed 2021-02-23 by rkaczorek