rkaczorek/astroberry-server #116: Failed to connect to server

capape, 2020-12-28:
Hi,

I am new to astroberry, congratulations for this product.
I have a problem with web access. When I click on connect button I am always getting "Failed to connect to server".

I installed astroberry from a debian buster installation. I could connect through vncviewer without problems, and I have used it through Skychart and ekos remotely from a laptop.

I am using my local wifi connection.

I don't where to look for logs where I can find out some hints to solve this problem.

Someone with this problem.

Thank you in advance,
Antonio.

rkaczorek, 2021-01-04:
Does it mean that you installed it on top of existing system using apt?
Make sure that nginx is running and upstream services are available as indicated in /etc/nginx/sites-enabled/astroberry

capape, 2021-01-05:
Thank you for your response.
Yes, I mean I installed astroberry on top of existing system. I have all the services included in that file up and running. I think that only desktop is failing. 
It seems something is closing websocket connection.

/var/log/astroberry/novnc.log:
...
localhost - - [05/Jan/2021 18:44:27] 127.0.0.1: Plain non-SSL (ws://) WebSocket connection
localhost - - [05/Jan/2021 18:44:27] 127.0.0.1: Version hybi-13, base64: 'False'
localhost - - [05/Jan/2021 18:44:27] connecting to: localhost:5900
localhost - - [05/Jan/2021 18:44:28] code 400, message Bad HTTP/0.9 request type ('\x82\x8c_\xeb.\x91')

Browser console:
* Error en el mapeo fuente: Error: request failed with status 404
  URL del recurso: http://192.168.1.142/static/js/bootstrap-select.min.js
  URL del mapa fuente: bootstrap-select.min.js.map
* Failed when connecting: Connection closed (code: 1000, reason: Target closed) rfb.js:668:21
    _fail http://192.168.1.142/desktop/core/rfb.js:668
    RFB http://192.168.1.142/desktop/core/rfb.js:210
    onclose http://192.168.1.142/desktop/core/websock.js:199

And the first attempt it shows this
Failed when connecting: Unsupported security types (types: 13,5,6,130,192) rfb.js:668:21
    _fail https://192.168.1.142/desktop/core/rfb.js:668
    _negotiate_security https://192.168.1.142/desktop/core/rfb.js:875
    _init_msg https://192.168.1.142/desktop/core/rfb.js:1269
    _handle_message https://192.168.1.142/desktop/core/rfb.js:713
    RFB https://192.168.1.142/desktop/core/rfb.js:186
    _recv_message https://192.168.1.142/desktop/core/websock.js:278

Best regards.

capape, 2022-02-13:
Latest versions are working

closed 2022-02-13 by capape