rkaczorek/astroberry-server #117: How to get INDI Library v1.8.8 (more specifically New ActiveFocuser driver release (#1301))

AnakChan, 2021-01-10:
Is there a way to upgrade Astroberry to INDI Lib v1.8.8? More specifically I'm interested in the new ActiveFocuser driver release (#1301), so if that works with the current Astroberry INDI Lib I'm also fine with that. I looked at :-

https://indilib.org/focusers/activefocuser.html

and tried sudo add-apt-repository ppa:mutlaqja/ppa but encountered an error presumably (is this an Ubuntu vs Raspbian thing?)  : 
astroberry@astroberry:~ $ sudo add-apt-repository ppa:mutlaqja/ppa
Traceback (most recent call last):
  File "/usr/bin/add-apt-repository", line 95, in <module>
    sp = SoftwareProperties(options=options)
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 109, in __init__
    self.reload_sourceslist()
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 599, in reload_sourceslist
    self.distro.get_sources(self.sourceslist)    
  File "/usr/lib/python3/dist-packages/aptsources/distro.py", line 93, in get_sources
    (self.id, self.codename))
aptsources.distro.NoDistroTemplateException: Error: could not find a distribution template for Raspbian/buster

Cheers,
Sean

marcocipriani01, 2021-01-14:
You don't need to add the PPA, it should be already set-up in the Astroberry (at least in mine, it is - Astroberry image downloaded a couple of months ago). Just running `sudo apt update && sudo apt full-upgrade` should bring you the latest INDI release.

AnakChan, 2021-01-14:
I've been running that apt update/upgrade/dist-upgrade a few times this week now however it hasn't pulled the new ActiveFocuser drivers down to Astroberry yet. I believe the ActiveFocuser were introduced only in INDI 1.8.8 which, itself, was released only a few days ago.

I guess the new INDI version releases aren't in sync with Astroberry yet, pending on Radek to incorporate it to the Astroberry build.

marcocipriani01, 2021-01-14:
A further investigation revealed that Astroberry's INDI release is supplied by its own [repository](https://www.astroberry.io/repo/) and not by `mutlaqja/ppa`. I think you have to wait for the new version to get there. If you really need the new driver now, you can compile INDI by yourself from the [sources](https://github.com/indilib/indi), which is a bit tedious.

AnakChan, 2021-01-21:
So I actually tried to compile the newest INDI v1.8.8 on the Pi itself (didn't have the environment for cross compilation), and whilst it actually did compile fine, I had issues with some of the drivers (e.g. ZWO didn't recognise my ASI cameras, Starbook crashed on start continuously, etc.).

It's been quite a few weeks since Radek/Astroberry maintainers been away to update their repository with the newest INDI.

marcocipriani01, 2021-01-21:
All I can tell you is to double check that:
- You installed every build dependency correctly
- You compiled a GitHub release and not the master branch (which is in development and may contain bugs)
- You are also compiling the [3rd party repository](https://github.com/indilib/indi-3rdparty), which contains the v1.8.8 ZWO ASI driver
   - You may need to install the ZWO SDK to properly compile the ASI driver - [link](https://astronomy-imaging-camera.com/software-drivers) (go to the "Developers" tab).
   - I have no idea on how to install it and if you really need it. I had to install the QHY SDK to compile the INDI QHY driver some time ago, but that's a different story.

Plus, you should report those bugs to the INDI developers.
Here's why I told you compiling INDI is tedious 🤣

jscheidtmann, 2021-02-15:
Astroberry runs the stable version of raspbian. V1.8.8 seems to be part of testing or unstable. Can't tell if the driver you're seeking is compiled for that. You can upgrade to that version or use pinning. See Debian docs: https://wiki.debian.org/DebianTesting or https://wiki.debian.org/AptConfiguration

jscheidtmann, 2021-02-15:
Or download indi files and dpkg -i, from here: http://archive.raspbian.org/raspbian/pool/main/i/indi/

jscheidtmann, 2021-02-15:
Use dpkg -l indi\* to list the installed packages.

AnakChan, 2021-02-16:
> Astroberry runs the stable version of raspbian. **V1.8.8 seems to be part of testing or unstable.** Can't tell if the driver you're seeking is compiled for that. You can upgrade to that version or [...]

https://www.indilib.org/forum/announcement/id-27.html
Not too sure where the the bolded comment is coming from. AFAIK INDI v1.8.8 is not beta nor test but a final version. About the driver I'm seeking for, it was introduced into v1.8.8 as per the ChangeLog.
+ New ActiveFocuser driver release (#1301)

Anyhow, I have managed to compile v1.8.8 and associated INDI 3rd party as per @marcocipriani01's directions. Therefore this issue can probably be closed.

closed 2021-02-16 by AnakChan

jscheidtmann, 2021-02-16:
Note: I was talking about the Linux distribution Astroberry is running on (raspbian, which is based on Debian). This has different "branches", that are named stable and testing. Astroberry is using stable, but the version you are looking for is in testing.

AnakChan, 2021-02-17:
I have to say I’m very confused about what you’re trying to explain. INDI is a cross-platform middleware that’s portable across multiple OSes. INDI v1.8.8 final release can be compiled and installed on any OS one chooses, may it be stable or unstable/test release.

As Radek hadn’t released an Astroberry package with INDI v1.8.8, I was asking how to upgrade the existing INDI v1.8.7 (that’s incorporated into Astroberry) to v1.8.8. That was asking for a final version a middleware product running on a stable RaspbianOS release. And since there was no progress in the Astoberry package 1 month after INDI v1.8.8 was released, I compiled it myself on the stable RaspbianOS Astroberry was running on.

jscheidtmann, 2021-02-19:
Yes, that's one way how to to it (compiling it yourself). I was explaining a different way, and that is installing v1.8.8 from another branch (testing) of the Linux distribution Astroberry is based on.

rkaczorek, 2021-02-22:
INDI 1.8.8 should be available in Astroberry  repo by the end of Ferbruary