rkaczorek/astroberry-server #119: AB and TP Link WLAN extender

gnnyman, 2021-02-10:
Sorry for asking this maybe too simple question... i wanted to extend the range of the built-in wlan by connecting the AB via LAN cable with my TP Link. The SSDIS is TP-Link_052E. It shows up in the WLAN list of the AB. Entering the PW of the TP Link worked as well. However, when I want to go online on the AB with the Chromium browser, I cannot open most websites and/or cannot downlaod anything. Has something to do with certificates... no idea what to do. 
Second problem is, that when I want to connect to the AB from my laptop, using the same TP Link, it does not connect to the graphical surface. 
Can you please give me some idiotproof advise, what to do? Thanks a lot!
Georg

jscheidtmann, 2021-02-19:
Please shutdown, attach the network cable, connect it to your TP Link, attach monitor via HDMI, keyboard and mouse and then boot. 

Open a console and copy and paste what's printed for 

$ ifconfig

Then what's printed by 

$ ping www.google.com

Last provide a screenshot of the 'error' screen of Chromium.

Thank you.

gnnyman, 2021-02-19:
Currently it rains, I get back to you, once I can use my equipment again…

 

Thanks,