rkaczorek/astroberry-server #120: Sidebar Function Problem

martini47, 2021-02-16:
I cannot move the sidebar with the hand symbol and the keyboard tab is not working. Everything else on the sidebar is fine, included the special keys. Not very important but still... Thanks for help!

jscheidtmann, 2021-02-19:
Are you accessing Astroberry from the browser? If yes, which browser? Please provide exact version info and the platform the browser is running on.

It generally helps, if you post the following info:
1) steps to reproduce: starting from common ground, describe how to get to the state, that's making problems. 
2) what you expect instead as behaviour.
3) other useful info, like versions or special equipment.

Thank you.

martini47, 2021-02-20:
Yes, with the browser. Mozilla Firefox 85.0.2 (64 bit) running on Win10 Pro 64bit. The problem is persistent and not changing with different security options under Firefox. Thank you!

rkaczorek, 2021-02-22:
What is the exact issue? Just to make sure that we are on the same side:
- hand symbol is used for screen panning on mobile devices, not for moving sidebar
- keyboard tab is for emulating special keys only, not to emulate entire keyboard

martini47, 2021-02-23:
I was not aware that hand symbol is only for mobile devices and only special keys are availlable. Thank you for your help and sorry for waisting your time, Martin

closed 2021-02-23 by martini47