rkaczorek/astroberry-server #121: Astroberry previous versions

drakth, 2021-02-19:
Where can i download astroberry previous versions? Like 2.0.2, My ASI290MM stopped working with the latest version.

rkaczorek, 2021-02-22:
Previous versions are not available for download. What is the problem with your ASI290MM?

drakth, 2021-02-22:
Astroberry does not recognize it. I found this post from someone who apparently has the same problem: https://www.indilib.org/forum/astroberry/7403-asi290mm-mini-not-recognized-any-more.html

vedrafal, 2021-02-23:
I'm sorry to inform you rkaczorek but I too have problems with the 2.0.3 and ASI533MC as well as EOSM50.
I think this is related to the same issue. 

What I could find out it looks like the problem is Debian/RaspiOS related cause it's out of date on some of the INDI libraries and Kstars/ekos. 
That's the reason I moved to Ubuntu Mate and am trying to rebuild the software I need from scratch. Both things work with the newest INDIlib and Kstars versions. 

I guess we just have to wait for the next big Debian release.

rkaczorek, 2021-02-23:
Have you tried to go through [this post](https://www.indilib.org/forum/astroberry/7403-asi290mm-mini-not-recognized-any-more.html#57503)?

drakth, 2021-03-30:
> Have you tried to go through [this post](https://www.indilib.org/forum/astroberry/7403-asi290mm-mini-not-recognized-any-more.html#57503)?

I have looked at that reply, but i dont know how to fully create the rule this person mentions. I have managed  to find a entry with an empty description but i dont know where to go from there, to create the custom rule.

rkaczorek, 2021-03-30:
Is it still valid issue? There were at least a few kernel updates in the meantime. Have you upgraded your system since then?

closed 2026-02-11 by rkaczorek