rkaczorek/astroberry-server #124: virtualgps not starting after dirty shutdown

luzik, 2021-02-25:
After power loose, VirtualGPS complying about /tmp/vgps and do not start. Removing that file fixes issue.

jscheidtmann, 2021-02-26:
Unfortunately this is a known problem with RaspPis. They need to be shutdown correctly, else issues can happen up to a crashed SDcard, so it doesn't boot anymore. Over time they got better, but still it can happen. 

What you can do is using some kind of battery and circuitry, that triggers shutdown, when power is lost. Or a press￼ button to shut it down before removing power, if you didn't use "shutdown".

For example I just ordered a StromPi and a battery hat, which includes an RTC, wide range power input, so it's a UPS. https://strompi.joy-it.net/ (hope it's in Englisch)

luzik, 2021-02-27:
What about removing that file on boot by Astroberry ?

rkaczorek, 2021-02-27:
Adding a procedure to remove the file at boot is just a workaround. tmp directory should not be persistent between reboots anyway.
Removing a file at virtualgps service start is not an option either, or it is... let me think about it.

rkaczorek, 2021-03-01:
It is already implemented. The issue you are facing is related to running virtualgps as root while the service uses gpsd user, which obviously does not have rights to remove socket file created by root.

closed 2021-03-01 by rkaczorek

luzik, 2021-03-02:
hmm what does it mean ?

$ service virtualgps status
● virtualgps.service - Virtual GPS
   Loaded: loaded (/etc/systemd/system/virtualgps.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2021-03-02 00:57:42 CET; 8s ago
  Process: 16194 ExecStart=/usr/bin/python3 /usr/bin/virtualgps.py (code=exited, status=1/FAILURE)
  Process: 16195 ExecStartPost=/bin/sleep 3 (code=exited, status=0/SUCCESS)
  Process: 16232 ExecStartPost=/usr/sbin/gpsdctl add /tmp/vgps (code=exited, status=0/SUCCESS)
  Process: 16233 ExecStopPost=/usr/sbin/gpsdctl remove /tmp/vgps (code=exited, status=0/SUCCESS)
 Main PID: 16194 (code=exited, status=1/FAILURE)

astroberry@astroberry:~ $ ls -l /tmp/vgps
lrwxrwxrwx 1 gpsd dialout 10 Mar  1 22:18 /tmp/vgps -> /dev/pts/0

$ sudo rm /tmp/vgps
astroberry@astroberry:~ $ service virtualgps status
● virtualgps.service - Virtual GPS
   Loaded: loaded (/etc/systemd/system/virtualgps.service; enabled; vendor preset: enabled)
   Active: active (start-post) since Tue 2021-03-02 01:02:04 CET; 597ms ago
 Main PID: 19352 (python3); Control PID: 19353 (sleep)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/virtualgps.service
           ├─19352 /usr/bin/python3 /usr/bin/virtualgps.py
           └─19353 /bin/sleep 3

autorestart not working for this service unless I remove /tmp/vgps It is not related to that file rights

rkaczorek, 2021-03-02:
I cannot reproduce this issue whatever I do. Please provide exact steps to reproduce it