rkaczorek/astroberry-server #126: oacapture failing to update

setnes, 2021-03-01:
Is this related to the new astroberry code release?

```
Unpacking oacapture (1.8.0-2) over (1.8.0-1) ...
Failed to restart udev.service: Unit udev.service is masked.
dpkg: warning: old oacapture package post-removal script subprocess returned error exit status 1
dpkg: trying script from the new package instead ...
Failed to restart udev.service: Unit udev.service is masked.
dpkg: error processing archive /var/cache/apt/archives/oacapture_1.8.0-2_armhf.deb (--unpack):
 new oacapture package post-removal script subprocess returned error exit status 1
Failed to restart udev.service: Unit udev.service is masked.
dpkg: error while cleaning up:
 new oacapture package post-removal script subprocess returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/oacapture_1.8.0-2_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

rkaczorek, 2021-03-01:
`Failed to restart udev.service: Unit udev.service is masked`
Why would you have udev.service masked on your system? What is the output of `sudo systemctl status udev.service`?

setnes, 2021-03-01:
```
$ systemctl status udev.service 
? udev.service
   Loaded: masked (Reason: Unit udev.service is masked.)
   Active: inactive (dead)
```

setnes, 2021-03-01:
unmasked and started.  oacapture installs.  The status now shows some errors with rules though.

```
? systemd-udevd.service - udev Kernel Device Manager
   Loaded: loaded (/lib/systemd/system/systemd-udevd.service; static; vendor preset: enabled)
  Drop-In: /etc/systemd/system/systemd-udevd.service.d
           ??override.conf
   Active: active (running) since Mon 2021-03-01 14:17:32 CST; 9s ago
     Docs: man:systemd-udevd.service(8)
           man:udev(7)
 Main PID: 28432 (systemd-udevd)
   Status: "Processing with 29 children at max"
    Tasks: 2
   CGroup: /system.slice/systemd-udevd.service
           ??  148 /lib/systemd/systemd-udevd
           ??28432 /lib/systemd/systemd-udevd

Mar 01 14:17:32 astroberry systemd[1]: Starting udev Kernel Device Manager...
Mar 01 14:17:32 astroberry systemd-udevd[28432]: /lib/udev/rules.d/52-PTR-2.rules:10: Invalid key/value pair, starting at character 34 (',')
Mar 01 14:17:32 astroberry systemd-udevd[28432]: /etc/udev/rules.d/90-svbonyusb.rules:5: Invalid key/value pair, starting at character 126 (',')
Mar 01 14:17:32 astroberry systemd-udevd[28432]: /etc/udev/rules.d/90-svbonyusb.rules:6: Invalid key/value pair, starting at character 94 (',')
Mar 01 14:17:32 astroberry systemd[1]: Started udev Kernel Device Manager.
~
```

rkaczorek, 2021-03-01:
I'm glad it fixes your issue with oacapture.
I have noticed the errors with udev too. I filed the bug report to the upstream repo. See [here](https://github.com/indilib/indi-3rdparty/issues/338)

setnes, 2021-03-01:
Thanks.

I don't know what caused udev to be masked like that.  I certainly didn't manually do it.

I did recently build INDI core, 3rd party libs, and drivers from source to test some GPSD fixes.  Does that install more than what is normally included with astroberry?  

The udev mask is odd, but I can't rule out this being self inflicted.

rkaczorek, 2021-03-01:
INDI core, 3rdparty libs and drivers are provided as they are, no changes to the upstream versions. Astroberry additionally provides other drivers, lots of system tweaks and useful astro apps.
udev was hard switched off, which must have been invoked manually. I don't think any app would do it. None of astroberry components for sure. Good it's working now because otherwise you would face lots of issues with your devices and drivers.