rkaczorek/astroberry-server #128: WIFI 5GHz

luzik, 2021-03-08:
I Was fighting weeks with unreliable 2.4 Wifi connection on PI4 4GB. 5Ghz never really worked, till day I switched from network-manager to manual wpa_supplicant.conf use. I believe this is related to setting country code for wifi driver, but network manager did not done this. Any way if someone struggling with bad wifi I won't tell you how to fix this, but as an workaround just do:
echo "options cfg80211 ieee80211_regdom=PL" > /etc/modprobe.d/cfg80211.conf
systemctl stop network-manager.service
systemctl disable network-manager.service
cp /etc/wpa_supplicant/wpa_supplicant.conf.bak /etc/wpa_supplicant/wpa_supplicant.conf
edit /etc/wpa_supplicant/wpa_supplicant.conf
and reboot
voila ! damnit - so much trouble gone !

jochym, 2021-03-08:
Just fill in *your* country code instead of PL!. Your problems @rkaczorek were probably rooted in the fact that some channels which are legal in US (Usually default setting) or China (also common setting) are excluded in other countries. You may be actually fined for using them! But more trivially - you are trying to transmit on the channel your AP is not using so you hop around and constantly switch channels. But this unused channel is so empty that it seems better then the other crowded ones, so you jump again etc... You get the picture. I have been there. Remember to fill in your regulatory zone. It often helps!

luzik, 2021-03-08:
The bad thing with network-manager is that You can set wifi regulatory zone using raspi-config, and those settings are never used because of network-manager controlling wifi.

With that issue fixed I can move to bad aligning with celestron-aux @jochym ;-)

luzik, 2021-03-08:
_Note that on Raspberry Pi devices that support the 5GHz band (Pi3B+, Pi4, CM4, Pi400), wireless networking is disabled for regulatory reasons, until the country code has been set. To set the country code, open the Raspberry Pi Configuration application from the Preferences Menu, select Localisation and set the appropriate code._

[source](https://www.raspberrypi.org/documentation/configuration/wireless/desktop.md)

jochym, 2021-03-08:
@luzik  *tionkuju aluzju poniał*, ;) as our eastern neighbours say. In particular, we are working on bugfixes and extensions to the driver. Here is the PR (WiP): https://github.com/indilib/indi-3rdparty/pull/348, Any help (comments, code, docs, testing) will be appreciated.

rkaczorek, 2021-03-08:
Raspi-config does not handle regional setting for Network Manager by design. It handles wpa_supplicant only - see its [source code](https://github.com/RPi-Distro/raspi-config/blob/master/raspi-config#L568). Regional setting with Network Manager can be handled by invoking wifi module with a parameter. So what you call a workaround is the actual solution `echo "options cfg80211 ieee80211_regdom=PL" > /etc/modprobe.d/cfg80211.conf`
I'm not aware of any GUI solution for doing this, but it is quite good idea to add it to next release of Astroberry Server.

pleiades-coram, 2021-03-13:
Hi luzik,

I don't know if it could help you but got very unstable connection with wifi since I got my Raspberry pi 4 B a month ago.

Finally, I think the problem was with the WPA2 security protocol.  Now I'm using WPA and everything seems working well.

Regards,
Yves