rkaczorek/astroberry-server #134: Add ability to set location from web page.

setnes, 2021-03-24:
This is an enhancement request to add a user interface that can populate the location.conf file from a web panel.

This would be very handy.  A web page can get location coordinates from the browser.  If using a device with GPS (almost any phone), the Astroberry coordinates could be set without needing to ever type them.

I have attached example client side code that fetches browser coordinates and displays text that could be used inside a location.conf file.  I did not attempt to write any server side code.

[location.html.txt](https://github.com/rkaczorek/astroberry-server/files/6193764/location.html.txt)

Getting the browser's location requires https.  This example html shows how to get and display the location.

I also know about NMEA.  This could be an alternative that doesn't require a phone app.

rkaczorek, 2021-04-06:
Thanks for reporting. This is good point

miceno, 2021-07-14:
👍