rkaczorek/astroberry-server #135: indi_rpicam crashes on PiZeroW

kairos10, 2021-03-31:
the indi_rpicam package installed from https://www.astroberry.io/repo/ crashes on a PiZeroW with an 'illegal instruction' error (base OS is Raspbian)
It might have something to do with the NEON instructions not being available on BCM2835.
Compiling the driver from source (on a pi4 and pushing the binary to the pi0 board) produced a similar error.
I was able to produce a working binary by changing CMakeCache.txt:NEON_FOUND:BOOL=false before 'make'

rkaczorek, 2021-03-31:
Thanks for reporting. Does it work on other versions of Raspberry Pi? If it's hardware dependent bug it must be handled accordingly with CMakeLists.txt. You could be interested @lboclboc

lboclboc, 2021-04-01:
I am interested, but sadly I don't have a PiZeroW to test on so I need to get one before I can give any feedback.