rkaczorek/astroberry-server #138: INDI 1.9.0 When?

etfloyd, 2021-05-01:
I know it's just newly released, but do you have an ETA yet for updating INDI to v1.9.0? Asking because would very much like to have a fix for this: https://github.com/indilib/indi/issues/1368.

rkaczorek, 2021-05-01:
I'm planning to push an update in a week or so, so stay tuned

SevenW, 2021-05-27:
Any updates on this? I am also very interested to move on to indi 1.9.0

JaimieMurdock, 2021-06-09:
Any update on this? I use the PMC-8, which received some substantial updates in this release and I want to see if it resolves an issue with middle-of-the-night crashes. I'm also interested in KStars 3.5.3, which doesn't seem to be in the repos yet either.

mr-wacha, 2021-07-03:
Just FYI in case someone skips on their system updates - INDI 1.9 seems now available from repos so just run:
`sudo apt update && sudo apt upgrade && sudo apt dist-upgrade`

 Seeing updated drivers for ToupTek/Omegon and kStars 3.5.3.

closed 2021-07-14 by rkaczorek