rkaczorek/astroberry-server #139: Astap Upgrades?

jscheidtmann, 2021-05-14:
It seems that astap is not present in the repositories and needs to be installed and upgraded manually. 

This should at least be reflected in the docs. 

ˋapt-cache showpkg astapˋ

gutaker, 2021-06-02:
AFAIK, ASTAP doesn't have its own repo and is installed manually, so in order to safely update, you need to remove it together with its database, then download newer version from [HNSKY site](https://www.hnsky.org/astap.htm) and install it manually.
Remove it with `sudo apt remove astap g17` or `sudo dpkg -r astap && sudo dpkg -r g17`
Install with `sudo dpkg -i astap_armhf.deb && sudo dpkg -i h18_star_database_mag18_astap.deb`

jscheidtmann, 2021-08-29:
I added it to the FAQ for the time being.

rkaczorek, 2021-08-29:
@jscheidtmann  thank you for your support. I have just did some minor edit to your [astroberry wiki](https://www.astroberry.io/docs/index.php?title=Astroberry_Server) entry.

closed 2026-02-11 by rkaczorek