rkaczorek/astroberry-server #140: My Experience and help Setting up KSTARS in Astroberry

bbillp, 2021-05-14:
Astroberry 2.03 or the latest download in March 2021
Raspberry Pi 4b 8gb memory and POWERED USB Hub for the Cameras.
previous Pi experience

No problems with WiFi from Windows PC out of the box
Added the $2.50 Real Time Clock from ‘the robot shop’ and followed OH8STN installation procedure.
Added the recommended U-Blox USB GPS following OH8STN procedure
         This GPS IS NOT recommended , poor sensitivity
INDI Drivers for GPSD and GPSNMEA, Celestron AVX mount, ZWO CCD camera (guide Camera) , Nikon DSLR for the Nikon D5600.
Firefox and VNCViewer for headless access

Desktop Size has to be adjusted in Raspi-Config separate for Firefox web browser and VNCViewr

Celestron AVX mount driver is ALSO CelestronGPS driver so I struggled to fix that then I read somewhere that the driver has 1 or 2 or 3 different names but are all the same.  anyway that work in Astroberry PHD2.

1) How is the ASTROBERRY password changed to something else ??????  I changed the Pi password but do not see how to change the astroberry password.

2) My big issue was setting up KSTARS, INDI drivers are setup and STarted in the INDI server (little arrow on the left margin of the desktop) I have KSTARS working to GoTo an object with the Celestron AVX mount but I am not sure HOW I got KSTARS configured for the drivers in the first place.  The Celestron AVX Nexstar + controller has a USB connection to the Pi-4 powered hub.  INDI driver is CelestronGPS which someone said is also Celestron AVX.  It seems that a cold reboot is required, I cannot be sure how I got KSTARS connected to the Celestron so Can anyone give me a clear cut 1,2,3, step setup.   

3) KSTARS astro map display is terrible, any hints to get the horizon and brighter DSS and Planets to appear?

Thanks to everyone who made this project possible, I wanted to dump my PC as a Astro-Controller and the Pi-4 with VNCViewer is a FAST combination.  My gear is outside remote controlled from the Kitchen and everything is battery powered.

Phylastik, 2021-05-17:
For question 1, yes i have the same issue, i was able to change the password for Pi (root user) and i used it for updates. But the password used for vnc is still the default one.

For question 2, it seems to be to more an issue with Kstars than with astroberry. So you should better have a look on the INDI librairy FAQ and forum to get more advices.

Question 3 : same thing

etfloyd, 2021-06-01:
1. From the astroberry desktop (I use the browser-based VNC desktop: https://<you pi's ip address>/desktop) just click the VNC icon in the upper-right corner just to the left of the signal strength indicator and a dialog box will pop up that gives you a link to change the VNC password: 
[image]
2. Using the Indi Web Manager (only available in the browser-based desktop afaik), I configure to start the indi drivers at astroberry start-up, then create a profile in Ekos and configure them as "Mode: remote" to localhost:
[image]
Having the indi drivers always running lets me connect directly from a remote kstars if I want. The other way to do it is not configure the indi drivers to start automatically on start-up, but set them up in Ekos as"Mode: local". That way they start when kstars starts and shut down when kstars shuts down. (If you're doing autoguiding with your AVX, you should know that the indi v1.8.9 Celestron GPS driver has a known issue. See: https://github.com/rkaczorek/astroberry-server/issues/138 for the link.)
3. Can't help you with this one. :(.

rkaczorek, 2021-07-14:
Regarding 3 - this needs to be addressed to KStars development team, not here.

bbillp, 2021-10-11:
Thank you for the replies.  The sky has been terrible so progress is very slow but PHD2 and Firecapture are working with a ZWO ASI120  for guiding and a ZWO ASI178 for imaging using the Pi USB-3 .  Also Celestron CGX and Celestron AVX are working fine.  NIKON DSLR Testing is next.

closed 2026-02-11 by rkaczorek