rkaczorek/astroberry-server #142: Problem with data transfer speed from 178MM to PHD2

parpocel, 2021-06-11:
Hi. I have some issue with ASI 178MM (lot of pixels - 6,4 MP) working on PHD2 in Astroberry. It's about refreshing time of looping. If I set exp time to 0,5sec I got real refreashing rate at twice that time. When I connect it to PHD2 installed on my more powerfull PC it's not an issue anymore. My question is, could it be fixed some how  by settings or it's just limitation of Raspberry (I have to say that I havent noticed overload on Pi at all - it operates ca 75 degree and 10-20% of CPU load. 2x2 binning helps but just slightly. Any ideas?

etfloyd, 2021-06-14:
I have the same camera in the same role. You'll probably have to configure PHD2 to use ROI while guiding to get below 1sec refresh rate. 
Under Advanced Settings:
  Camera tab - check Use Subframes
  Guiding tab - enlarge the Search Region to maybe 20-25 pixels so it won't lose the star between exposures
Using Subframes greatly reduces the data transfer size for each frame allowing much faster exposures. If it loses the star, it will revert to the full frame to find another one, then drop back to subframe. Note that when it's guiding, you'll only see the tiny subframe with the guide star on the screen. Check View, Display Star Profile to see a blow-up of the subframe in the cross-hairs.

Some might advise that 0,5 sec is too fast and you're "chasing the seeing". If you're reducing exposure purely to get a better (pointier) star profile, it may be better to reduce the gain and use a longer exposure. (FWIW, my mount seems to track better with short exposures too, so I usually ignore that advice.)

parpocel, 2021-06-16:
Yup. I realized, that it's just hardware limitation of Pi and I will try to guide on subframes on about 1-2sec. Thanks for the response.