rkaczorek/astroberry-server #144: Update issue

LondonAstro, 2021-07-11:
Problem was failed in 'sudo apt upgrade', had to do the following
sudo apt update
sudo apt upgrade
sudo apt upgrade --fix-missing
sudo apt dist-upgrade

Regards
Paul

mgutierrezp, 2021-07-21:
I'm also suffering problems with a completely new fresh install of astroberry 2.0.3 and apt upgrade commands. These are the very first commands I run:

# apt-update
[...]
Fetched 13.6 MB in 9s (1,564 kB/s)                                                                                                                                                                                                           
Reading package lists... Done

# apt-get upgrade -y
[...]
Setting up indi-mqtt (1.0.8-1) ...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyindi-client
  Downloading https://files.pythonhosted.org/packages/bd/2b/4b7c43b20e35ea838706a63c86fe321647c19bbc996c49971369b216d6e7/pyindi-client-0.2.7.tar.gz
Building wheels for collected packages: pyindi-client
  Running setup.py bdist_wheel for pyindi-client ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-rvkxhryw/pyindi-client/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-nu3ofqca --python-tag cp37:
  running bdist_wheel
  running build
  running build_ext
  building '_PyIndi' extension
  swigging indiclientpython.i to indiclientpython_wrap.cpp
  swig -python -v -Wall -c++ -threads -I/usr/include -I/usr/include/libindi -I/usr/local/include/libindi -o indiclientpython_wrap.cpp indiclientpython.i
  indiclientpython.i:37: Error: Unable to find 'indimacros.h'
  indiclientpython.i:43: Error: Unable to find 'indiwidgettraits.h'
  indiclientpython.i:44: Error: Unable to find 'indipropertyview.h'
  Language subdirectory: python
  Search paths:
     ./
     /usr/include/
     /usr/include/libindi/
     /usr/local/include/libindi/
     ./swig_lib/python/
     /usr/share/swig3.0/python/
     ./swig_lib/
     /usr/share/swig3.0/
  Preprocessing...
  error: command 'swig' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for pyindi-client
  Running setup.py clean for pyindi-client
Failed to build pyindi-client
Installing collected packages: pyindi-client
  Found existing installation: pyindi-client 0.2.5
    Uninstalling pyindi-client-0.2.5:
      Successfully uninstalled pyindi-client-0.2.5
  Running setup.py install for pyindi-client ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-rvkxhryw/pyindi-client/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-xlgo91q3/install-record.txt --single-version-externally-managed --compile:
    running install
    running build_ext
    building '_PyIndi' extension
    swigging indiclientpython.i to indiclientpython_wrap.cpp
    swig -python -v -Wall -c++ -threads -I/usr/include -I/usr/include/libindi -I/usr/local/include/libindi -o indiclientpython_wrap.cpp indiclientpython.i
    indiclientpython.i:37: Error: Unable to find 'indimacros.h'
    indiclientpython.i:43: Error: Unable to find 'indiwidgettraits.h'
    indiclientpython.i:44: Error: Unable to find 'indipropertyview.h'
    Language subdirectory: python
    Search paths:
       ./
       /usr/include/
       /usr/include/libindi/
       /usr/local/include/libindi/
       ./swig_lib/python/
       /usr/share/swig3.0/python/
       ./swig_lib/
       /usr/share/swig3.0/
    Preprocessing...
    error: command 'swig' failed with exit status 1
    
    ----------------------------------------
[... 85 lines cut ...]
5 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up gpsd (3.17-7+b1) ...
Creating/updating gpsd user account...
Setting up astroberry-server-wui (2.0.7-1) ...
Synchronizing state of nginx.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable nginx
Setting up astroberry-server-sysmod (2.0.5-1) ...
Setting up virtualgps (1.1.0) ...
Setting up astroberry-server-full (2.0.3-1) ...
root@astroberry:~# apt-get upgrade -y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  libtiffxx5 python-colorzero
Use 'apt autoremove' to remove them.
The following packages have been kept back:
  indi-bin indi-full indi-pentax kstars-bleeding kstars-bleeding-dbg libindi-dev libindi1 oacapture raspberrypi-sys-mods raspberrypi-ui-mods
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.

After the (incomplete) upgrade, it results in a weird behaviour on phd2 (I cannot click the 'connect' button in Indi dialog to configure the guidecam, although I don't know if it is related at all, but if I don't apt-get upgrade, phd2 works perfectly)

cheers

bbillp, 2021-07-21:
Don’t perform the upgrade, 2.0.3 limps without them.  No problems running PHD2 with Celestron AVX & CGX, imaging with Nikon 5600, ZWO ASI178 guiding.    I find it easier to manually control the imaging camera than trying to figure out KSTARS and EKOS.      Bill P.

mgutierrezp, 2021-07-21:
this is very strange. I've done another test. Installed again from scratch. I did only upgrade PHD2, from 2.6.9 to 2.6.10, and the weird behaviour described above, appears again. PHD2 2.6.9 works perfectly, but when upgraded to 2.9.10 I cannot connect to the guidecam (exposed by the indi server on localhost). When I click on phd2 'connect' button, absolutely nothing happens; although I see error messages (cannot connect) in phd2 logs. I've indeed opened a ticket to phd2 developers **but** I really think it is not a bug of phd2. I have another phd2 2.6.10 installation on my linux desktop (linux mint), and can connect to the indi server in the rpi. Maybe a bug in the phd2 .deb packaging? any ideas?

LondonAstro, 2021-07-21:
Just to add at the time of the upgrade I got PHD2 2.6.9dev 4, which I was happy with. Plus like Bill I do not define my guide camera (Orion SSAG) in the Indi Server, but directly point PHD2 to it (defined as QHY Camera). The Orion SSAG has been native support in PHD2 for a number of years, but there is definitely some issue with the basic upgrade.

rkaczorek, 2021-07-22:
@mgutierrezp most problems you are reporting are related to indi-mqtt which depends on extrenal library pyindi-client (python bindings for INDI). It has became unstable recently and the author decided to announce end of life of this library. If you don't use MQTT uninstall indi-mqtt before upgrade. The second problem is related to gpsd. To fix it run the following sequence of commands:
```
sudo systemctl stop gpsd.service
sudo systemctl stop gpsd.socket
sudo dpkg --configure -a
sudo systemctl restart gpsd.service
sudo systemctl restart gpsd.socket
```
I will be releasing clean and updated version 2.0.4 soon. It's a long time since 2.0.3 has been released

LondonAstro, 2021-07-22:
Great response, I must say how impressed I am with Astroberry, I've been into Astro photography for about 8 years (Windows based) and find this solution easy to install and use. I assume I'm allowed to close the issue....

closed 2021-07-22 by LondonAstro

mgutierrezp, 2021-07-22:
the workaround described above has worked perfectly. I suspected it was related with ind-mqtt. PHD 2.6.10 is still not working (same behaviour as described). Version 2.6.9 works. Do not know where the problem really is

bbillp, 2021-08-01:
What are the old and updated Astroberry version numbers ?

rkaczorek, 2021-08-01:
Old is all below 2.0.3 however packages are updated within the unchanged version. Think of it as updating windows. All fixes and patches do not change windows 10 to anything else.

bbillp, 2021-08-01:
Thanks for the rapid explanation.  Can you give me the command line to properly update?  My version is about 6 months old.

rkaczorek, 2021-08-02:
Run ordinary update command i.e. `sudo apt update && sudo apt upgrade`
If there is any issue with the update run the sequence:
```
sudo systemctl stop gpsd.service
sudo systemctl stop gpsd.socket
sudo dpkg --configure -a
sudo systemctl restart gpsd.service
sudo systemctl restart gpsd.socket
```
Then you can run updates without problems

bbillp, 2021-08-02:
Thank you, I have learned from 6 years Pi experience never to update unless its extremely necessary as I’ve lost use of some good astronomy apps.   I’ll update a backup SD …. Thanks for the info.