rkaczorek/astroberry-server #145: Astroberry-2.0.3 fails on new Pi4

khobar95, 2021-07-29:
I installed Astroberry-2.0.3 on a Pi4B 4GB system I bought late last year, and it is working fine. I bought another Pi4 4GB to run the HQ camera. However, I decided I wanted to set up another Astroberry-2.0.3 system. This second system has major issues. It either fails to boot, boots to the Astroberry background pic, or boots all the way and then locks up for several minutes if I move the mouse. I put the microSD card into my "working" Pi and it booted no problem, in seconds. This was with the Pi connected to a monitor, keyboard, and mouse.

I thought perhaps hardware issue, so I burned raspiosbuster-full, and it works just fine.

I have tried various microSD cards, and the results are the same. I am at a complete loss to understand what could be going wrong. 

Thoughts?

Thanks!

rkaczorek, 2021-07-29:
Have you updated your system after flashing or it is vanilla fresh from the image?

Mucephie, 2021-07-29:
I'm getting the exact same issue with the same conditions (although I'm using an 8GB board). This occurs for me for a vanilla install, I have managed to get as far as doing the recommended "sudo apt update && sudo apt upgrade && sudo apt dist-upgrade" but once it reboots the issue persists.

Mucephie, 2021-07-29:
Based on the mention of an HQ camera in this post , the fact  I have an HQ cam plugged in, and  issue #131  Having a similar situation. I think its safe to say that the interaction between astroberry and the HQ cam is the most likely culprit.

Mucephie, 2021-07-30:
Just did a fresh install test with the HQ cam unplugged, and although it took some time it did eventually boot to the desktop. I tried enabling the camera interface in raspi-config and rebooting to see if that broke things, and it didn't. Ill try reconnecting the camera next to see if having the camera on first boot breaks things, or if the camera being present breaks any boot attempt.

Mucephie, 2021-07-30:
Sorry to spam this thread, but I did get it to boot properly with the Pi HQ cam connected. It appears that it cannot be connected during first boot, but can be on subsequent boots (I'm not sure if enabling the picam interface in raspi-config is required before connecting). As for whether the PiCam HQ is usable after this is unknown since I still don't know how to connect it in software to astroberry (some help would be appreciated).
@khobar95 Try running the Pi without the camera attached, once the first boot sequence completes enable the picam interface in raspi-config and you can connect the camera and booting should progress as normal. I hope this resolves your issue

rkaczorek, 2021-07-30:
So it looks like it is hardware related.
@Mucephie you need to use indi_rpicam driver to access the camera.

khobar95, 2021-07-31:
I am running the Pi without the camera attached. 

The "problem" Pi4 works flawlessly with raspian (raspios-buster). It does not reliably boot with vanilla 2.0.3, but it does come up *if* some updates have been applied using a different Pi. I discovered this when I used a PiZero, but it got to 96% before the Zero became unresponsive. However, rebooting the Zero, Astroberry came up. Hmm... So, I put the SD card into the "problem" Pi, and it booted to Astroberry. I thought maybe I could redo the update/upgrade. It failed.

I brought in my "working" Pi and have been using it for testing all day. I flashed a new image, and it boots. So far, so good. However, when I attempt to upgrade using sudo apt update && sudo apt upgrade && sudo apt dist-upgrade, it eventually comes back with the same errors as the PiZero and the "problem" Pi such as:

Jul 31 05:38:04 astroberry systemd[1]: gpsd.service: Job gpsd.service/start failed with result 'dependency'.
Jul 31 05:38:06 astroberry systemd[1]: Dependency failed for GPS (Global Positioning System) Daemon.
Jul 31 05:38:06 astroberry systemd[1]: gpsd.service: Job gpsd.service/start failed with result 'dependency'.
dpkg: error processing package gpsd (--configure):
 installed gpsd package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of astroberry-server-wui:
 astroberry-server-wui depends on gpsd; however:
  Package gpsd is not configured yet.

dpkg: error processing package astroberry-server-wui (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of astroberry-server-sysmod:
 astroberry-server-sysmod depends on gpsd; however:
  Package gpsd is not configured yet.

dpkg: error processing package astroberry-server-sysmod (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of virtualgps:
 virtualgps depends on gpsd; however:
  Package gpsd is not configured yet.

dpkg: error processing package virtualgps (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of astroberry-server-full:
 astroberry-server-full depends on astroberry-server-wui; however:
  Package astroberry-server-wui is not configured yet.
 astroberry-server-full depends on astroberry-server-sysmod; however:
  Package astroberry-server-sysmod is not configured yet.
 astroberry-server-full depends on virtualgps; however:
  Package virtualgps is not configured yet.

dpkg: error processing package astroberry-server-full (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 gpsd
 astroberry-server-wui
 astroberry-server-sysmod
 virtualgps
 astroberry-server-full
E: Sub-process /usr/bin/dpkg returned an error code (1)

There are some more similar messages before this, too.
I checked the checksum sha256 and it's correct. 

I hope this helps!

Thanks!

rkaczorek, 2021-07-31:
See this issue #144

bbillp, 2021-07-31:
In my experience the SD built and booted on one pi will not necessarily work on another pi.  I learned the hard way so I always make fresh SD cards for each individual Pi here models 1-4 .     BP

khobar95, 2021-07-31:
Thank you Radek - that explains a lot! I noticed last night when I installed gpsd and clients (sudo apt-get install gpsd gpsd-clients), there were references to some of what was previous failing (wui, for instance). 

I am unsure of the actual state of my Pi4's now, but the "problem" Pi4 is booting and stable. I will keep an eye on things.

Again, thank you.