rkaczorek/astroberry-server #146: Password Issues

ccx004B, 2021-08-01:
When I boot the pi running astroberry it asks for my password three times in succession. When I try to run  preferences/Raspberry Pi Configuration it asks for my password three times and although I am typing the correct password, it never starts. If I type:

sudo passwd root

in a shell it responds "[sudo] password for astroberry". It does let me type a password but no matter what I type it responds "Sorry, try again."

What on earth is going on?

ccx004B, 2021-08-01:
Sorry, I just realised that the password is the default one not the one I set or at least I thought I set when I installed it.

closed 2021-08-01 by ccx004B

rkaczorek, 2021-08-01:
The command you tried sets password for root account, not astroberry. If you want to change default password for astroberyr you should just run `passwd`

ccx004B, 2021-08-01:
Hi Radek,

 

Thanks for the info. Yes, as soon as I had posted the query I realised what I was doing wrong.

 

Best wishes….

Colin