rkaczorek/astroberry-server #147: Welcome to nginx! - no browser access

M31Fan, 2021-08-13:
Hello! 

I'm hoping this is an easy fix, but Linux is not my strength. 

After an update, which didn't present any error, I had a notification that MANY packages were no longer used, and it recommended to "autoremove".... and halfway through it, the desktop and its windows were replaced with a "welcome to nginx!"

I'm still able to access the desktop but only through a VNC application. The Web browser is stuck on that message. 

Any recommendations to fix and avoid this in the future? 

Many Thanks!
Tiago

M31Fan, 2021-08-13:
Fixed! 
I accessed the RPi 4 through the VNC and installed several "astroberry" packages, one of them with the WUI, which I think it was the relevant one. 

Any change to make these packages not being removed when we use the autoremove feature that is suggested at the end of an update? The autoremove freed over 800 Mb of space, so it's good to be able to recover the space, but not good to loose the WUI in the process. 

Many Thanks!
Tiago

rkaczorek, 2021-08-14:
This comes from the fact that you removed astroberry-server-full at some stage, This package does not provide anything on it own and only installs all Astroberry components for your with just one command. While it is uninstalled all packages that it installed are treated as orphans so `sudo apt autoremove` just sweeps them out. To avoid it you can either:
- reinstall it by running `sudo apt install astroberry-server-full` or
- install all components directly using [this](https://github.com/rkaczorek/astroberry-server#where-is-the-source-code-of-the-system) list

If you ran  `sudo apt autoremove` you can only use the first strategy to recover your system