rkaczorek/astroberry-server #148: Astroberry Indi ZWO 294

Rory-Bugs, 2021-08-31:
Trying to upgrade to astroberry.  Long time user of RPI B+ with Mint for remote control of telescope and gear.  All works well with astroberry except the server won't work with my ZWO 294 Color Pro camera.   The Oacapture does recognize the camera as does indistart installed on other various Linux laptops.   In other words I'm loading ccd zwo in the web server for a driver but Kstars and CCDCIEL don't connect.  AVX mount and Starshoot guider working well.  Tried to update to latest indi-asi but I had the latest installed.  I installed indistart but it had the same problem  Some insight would be appreciated.  Thanks

rkaczorek, 2021-08-31:
Did you run upgrade after installing from image?

Rory-Bugs, 2021-08-31:
> Android [4].
> Links:

Yes multiple times.  Here's some output that may point to the problem.
The following packages have been kept back:
   indi-bin indi-full indi-pentax kstars-bleeding kstars-bleeding-dbg
   libindi-dev libindi1 oacapture phd2 raspberrypi-sys-mods 
raspberrypi-ui-mods
0 upgraded, 0 newly installed, 0 to remove and 12 not upgraded.
5 not fully installed or removed.

dpkg: error processing package gpsd (--configure):
  installed gpsd package post-installation script subprocess
  returned error exit status 1

Then later there's this

Errors were encountered while processing:
  gpsd
  astroberry-server-wui
  astroberry-server-sysmod
  virtualgps
  astroberry-server-full
E: Sub-process /usr/bin/dpkg returned an error code (1)

Hope this helps out.  Thanks again

rkaczorek, 2021-09-01:
Run clean updates first. Make sure to read [this](https://github.com/rkaczorek/astroberry-server/issues/144#issuecomment-884674509).

closed 2021-09-07 by Rory-Bugs