rkaczorek/astroberry-server #153: sudo command error

jLadd42, 2021-09-09:
When trying to flash Astroberry Image file using "sudo dd if=astroberry-server_2.0.3.img of=/dev/Astroberry bs=8M status=progress" I get the Terminal error message "dd: bs: illegal numeric value".  Copy/Paste the terminal error message from the Terminal gets:

[image]

Being a noob to Astroberry, Github, etc. I have no clue what the last means.  How do I fix the bs=8M numeric value?

Thanks

rkaczorek, 2021-09-09:
On what system are you running this command?

jLadd42, 2021-09-09:
My bad, thought I included that: 2019 iMac desktop using Terminal (V2.11(404)) running ZSH.

rkaczorek, 2021-09-10:
I'm not mac user so I will not help too much here. Anyway you can actually skip the last two parameters and run just this:
`sudo dd if=astroberry-server_2.0.3.img of=/dev/sdX`
or you can use any gui tool to flash SD card e.g. Raspberry Pi Imager from [here](https://www.raspberrypi.org/software/)