rkaczorek/astroberry-server #163: EqmodGui

GrabzIt, 2021-10-31:
Hi, I am loving the new version of astroberry. I was just investigating eqmod as I need it for a new piece of equipment. However, from what I can see it is missing a graphical user interface in astroberry. This seems to be provided by: https://sourceforge.net/projects/eqmodgui/

The same developer already has created the excellent Skycharts and CCDCiel.

Please could this be added to the astroberry repository? As it will make eqmod much easier to configure.

rkaczorek, 2021-10-31:
You can download and install it directly from https://sourceforge.net/projects/eqmodgui/files/eqmodgui_1.7.0/
For raspberry Pi you should use armhf version i.e. eqmodgui_1.7.0-127_armhf.deb
Anyway you can also use similar graphical user interface for your mount, which is part of KStars/Ekos. Go to mount tab and click Mount Control button (top right).

GrabzIt, 2021-10-31:
Ah thanks I will do that. I use Ccdeil for my imaging as I find it works better on astroberry. Unfortunatley the developer separated this feature out from the main app.