rkaczorek/astroberry-server #164: WIFI stop working when connecting SSD to USB3

ilkanaev, 2021-11-09:
astroberry-server version 2.0.4
raspberry pi 4 with 4GB
Kingstone 120 Gb SSD disk with USB3 adaptor

Perhaps this issue related to https://forums.raspberrypi.com/viewtopic.php?t=251487?

Scenario:
Step #1 connect astroberry to home network
Step #2 connect SSD to USB3 port on raspberry PI (there is no any issue with USB2 ports, only connecting to USB3 brings this issue)

Result: 
The WIFI stop working.
After reboting the astroberry with SSD connected to USB3, it will switch to HotSpot mode and will not be able to connect to home network.

bbillp, 2021-11-09:
Could this be a power problem or the Hard Drive electronics creating substantial radio frequency interference to the WiFi ????

ilkanaev, 2021-11-10:
Hi,
Please note, there is no any issue when i am using usb2 port. Only when
using usb3 port on raspberry.

Thanks
Yossi

rkaczorek, 2021-11-10:
It it happens only for USB 3.0, I'm almost sure it is well known problem of interference between USB 3.0 and WiFi 2.4GHz.
Try connecting to WiFi 5Ghz network instead and see if it works. If it does not, your SSD is power hungry and your raspberry probably faces undervoltage issue. You can verify this by observing system logfile for a while `tail -f /var/log/syslog`
Powering SSD with external power supply would be good idea anyway.

ilkanaev, 2021-11-11:
Thank you for quick and detailed responses!!!
I switched to use the same SSD for booting the device instead of using the
SD card  (set astroberry image directly on SSD ) and now all working fine.

Thank you very much!

closed 2021-11-11 by rkaczorek

ilkanaev, 2021-11-11:
Same ssd, same raspberry and same power supply....