rkaczorek/astroberry-server #165: Minimal exposure time is 1/10 for Sony DSLR in INDI and in EKOS

ilkanaev, 2021-11-16:
Hi,
i am running with astroberry 2.0.4 on raspbery pi 4 (4Gb).
i have Sony a6500 DSLR as my main camera. The camera set to Manual -> BLUB mode.
The camera connected to the D-LINK USB3 hub (with few other devices), and the hub connected to the USB3 port of rasbery.

When i am using other software like APT or Sharpcap, i am not using BLUB mode and the exposure on my camera is fully controlled by the SW. The APT/Sharpcap actualy updateting the exposure on the camera itself and waiting for the image to be received.
In case of INDI, looks like the only option to control the exposure is using BLUB mode. 
In my case, i can set/select any exposure time in INDI and in EKOS, but the actual minimal exposure is alwasy limited by 1/10 sec. and i can not get anything less than this.

If i swith the camera to any but not BLUB exposure, the exposure defined directly on my camera will be used, and no any way to change it from INDI/EKOS.

If i switch the camera to BLUB, I can enter/select any vaues for the exposure in INDI [DSLR tab]->[Main Control]->[Presets or Expose], but the actual exposure time will not go down than 1/10.

If i disable the BLUB mode in INDI, and select any exposure, it looks like during taking the picture the Sony DSLR driver used by INDI is crashing and the software loosing connection to the camera (perhaps camera doing reset, i was not able to understand that ) . The only way to recover from this situation is by clicking on 'restart driver' option next to the camera name in EKOS.

i saw several posts on this (mostly dealing with Flat frames, and enabling/disabling BLUB settings in INDI), but noone of these helped me to get the actual exposure time controled by INDI and lower than 1/10.

i placed the same issue at https://indilib.org/forum/ccds-dslrs/10721-minimal-exposure-time-is-1-10-for-dslr-on-indi-with-ekos-astroberry-image.html

Any idea how this could be resolved? if not, where i can ask for help on such issues?
Astroberry is GREAT (!!!) package, i really love it, and would like to use it for the entire astrophotography flow. but as per now Flat frames is not an option and photographing Moon and Jupiter as well :-(.

Thank you in advance,
Yossi

rkaczorek, 2021-11-17:
Hi Yossi. Thanks for reporting.
I'm afraid I will not be able to help much with this issue as it is INDI driver specific. The best place to file this issue is official INDI repository https://github.com/indilib/indi-3rdparty

However... Assuming you use indi-gphoto driver for your camera you can test low level library used by this driver i.e. libgphoto.
If you can control your camera with gphoto2 command from command line and achieve your goals, it means that it is a driver problem. If you cannot get shorter exposures using gphoto2 command, it means it is library problem and INDI team, including myself cannot do much about it. So the solution would be to wait for a new version of libgphoto - good news is it is provided by the latest Raspberry OS. Try it out and report back if you find out anything interesting.

ilkanaev, 2021-11-17:
Hi.
Thank you for quick response!

i did as following, and as result got the image with right exposure. 

1. installed the gphoto2 , by  **_sudo apt-get install gphoto2_**
2.  executed **_gphoto2 --auto-detect_** and got in result
**_astroberry@astroberry:~ $ gphoto2 --auto-detect
Model                          Port
----------------------------------------------------------
Sony Alpha-A6500 (Control)     usb:001,008_**

3. Executed **_gphoto2 --get-config /main/capturesettings/shutterspeed_** and got in result

**_astroberry@astroberry:~ $ gphoto2 --get-config /main/capturesettings/shutterspeed
Label: Shutter Speed
Readonly: 0
Type: RADIO
Current: 250/10
Choice: 0 30
Choice: 1 25
Choice: 2 20
Choice: 3 15
Choice: 4 13
Choice: 5 10
Choice: 6 8
Choice: 7 6
Choice: 8 5
Choice: 9 4
Choice: 10 32/10
Choice: 11 25/10
Choice: 12 2
Choice: 13 16/10
Choice: 14 13/10
Choice: 15 1
Choice: 16 8/10
Choice: 17 6/10
Choice: 18 5/10
Choice: 19 4/10
Choice: 20 1/3
Choice: 21 1/4
Choice: 22 1/5
Choice: 23 1/6
Choice: 24 1/8
Choice: 25 1/10
Choice: 26 1/13
Choice: 27 1/15
Choice: 28 1/20
Choice: 29 1/25
Choice: 30 1/30
Choice: 31 1/40
Choice: 32 1/50
Choice: 33 1/60
Choice: 34 1/80
Choice: 35 1/100
Choice: 36 1/125
Choice: 37 1/160
Choice: 38 1/200
Choice: 39 1/250
Choice: 40 1/320
Choice: 41 1/400
Choice: 42 1/500
Choice: 43 1/640
Choice: 44 1/800
Choice: 45 1/1000
Choice: 46 1/1250
Choice: 47 1/1600
Choice: 48 1/2000
Choice: 49 1/2500
Choice: 50 1/3200
Choice: 51 1/4000
Choice: 52 1/5000
Choice: 53 1/6400
Choice: 54 1/8000
Choice: 55 1/10000
Choice: 56 1/12500
Choice: 57 1/16000
Choice: 58 1/20000
Choice: 59 1/25000
Choice: 60 1/32000
Choice: 61 Bulb
END_**

4. Executed  **_gphoto2 --set-config-index /main/capturesettings/shutterspeed=40_**
5. Executed  **_gphoto2 --capture-image-and-download_**

by the end i got an image with exposure = 1/3200 (as expected)
So from my understanding, it is a driver problem.

What to do next?

Thank you in advance,
Yossi

rkaczorek, 2021-11-18:
Please provide output from `gphoto2 --version`

closed 2021-11-18 by rkaczorek