rkaczorek/astroberry-server #166: Default to WiFi connection

petro62, 2021-12-03:
I recently set my astroberry up and things seem to be running pretty good.  For some reason though I can't get it to connect to my WiFi network on reboot.  On every reboot I log into the hotspot then I have to go to 'Connect to Hidden Network' and then from there in the drop down list I see my network and connect.  I am not sure why I have to got to the hidden network option.  I also noticed that that no other networks show up when I connect to the hotspot and click the Signal symbol in the upper right.  All I see is the astroberry hotspot.

I tried moving the priorities around, but I feel like it may be something else since I have to do the whole 'connect to hidden network'

Also does all the software in astroberry auto update or will I have to go check every so often?

jscheidtmann, 2021-12-04:
A common problem is that the WiFi driver needs a WiFi country to correctly enable some features. So set that in raspi-config and check it‘s present in /etc/default/crda (FAQ 8.5). Else At least I am out of my depth and you may find more info in wpa_supplicant related posts, as that is what Astroberry is using…

petro62, 2021-12-05:
I thought I set that up after reading that, but I will double check to make sure.  I changed in raspi config, but I will need to check the other spot you mentioned.

petro62, 2021-12-14:
I double checked my settings and went to the path you mentioned,but it said I did not have permission to change.  So far I still have to connect via the hotspot then set it to connect to the wifi and then reconnect through wifi.  It isn't a big deal, but if I could eliminate a step it would be nice.

SlackJ, 2021-12-16:
`sudo nano /etc/default/crda` in the terminal will let you edit that file. You'll be prompted for your password after entering that command.

petro62, 2021-12-20:
> `sudo nano /etc/default/crda` in the terminal will let you edit that file. You'll be prompted for your password after entering that command.

Working good now.  Thanks

closed 2021-12-20 by petro62