rkaczorek/astroberry-server #167: File Folders to Backup

bbillp, 2021-12-03:
Which are the Allsky and system file folders names to backup to external HD?

jscheidtmann, 2021-12-04:
Settings and even astrometry data files are stored in /home/Astroberry/.local, .indi, .gphoto, etc. and /etc so given your backup config you should be done. The litmus test of every backup is doing a successful restore: So practice that. Do this at your own risk. You have been warned.

The recommended way - I believe - is to clone the SDcard using the SD copy tool available from the menu. Plug-in a SDCard reader and a new SDCard, then copy the whole thing over. The pro of that is, that you‘ll be ready to continue out in the field, if the card physically breaks and you‘ll only loose some data and not the night. (And it will some day, given Murphy‘s law).

jscheidtmann, 2021-12-04:
Also check: https://github.com/rlancaste/AstroPi3/blob/master/backupOrRestore.sh

bbillp, 2021-12-05:
Thank you for the information on the backup..  /home has been backed up on this Ubuntu system while backup SD cards have secured the Raspberry Pi system.    tnx

closed 2021-12-05 by bbillp