rkaczorek/astroberry-server #169: It's quite important thing to note that you use KStars on Windows 10 indeed. However, if so, your question should be "how to install astrometry on Windows 10", instead "how to install all astronometry.net index files on astroberry". The answer to the former question is, astrometry is not supported on Windows. The answer to the latter question has been already provided above.

Lvzyla, 2021-12-09:
It's quite important thing to note that you use KStars on Windows 10 indeed. However, if so, your question should be "how to install astrometry on Windows 10", instead "how to install all astronometry.net index files on astroberry". The answer to the former question is, astrometry is not supported on Windows. The answer to the latter question has been already provided above.
Also please note that I do not provide support for KStars and Windows here. It is Astroberry Server project. If you follow my step-by-step instructions, use KStars running on Raspberry Pi and connect to it either via web browser or VNC, you will find out there is no issue at all.

_Originally posted by @rkaczorek in https://github.com/rkaczorek/astroberry-server/issues/51#issuecomment-634676817_

Lvzyla, 2021-12-09:
Its Dec of 2021. I have been using Kstars/Ekos on Win 10 with the latest version. I connect to a RPI4 running Astroberry and acting solely as sn Indi server. Is it still truethat one is unable to plate solve on Win 10 Kstars whether using stellar solver or Astap installed on Win 10. I tried it the other nightsnd get getting error messages. Never did see Astap doing anything.
Lou Z.

closed 2021-12-09 by Lvzyla

rkaczorek, 2021-12-09:
Could you change title to anything meaningful?

rkaczorek, 2021-12-09:
BTW you can always plate solve using online mode