rkaczorek/astroberry-server #170: Atik 320e problem

Paul47910106, 2021-12-14:
Hi there.
I've run Astrobrry for a little while now and find it to be very good and easy to navigate as a complete novice. Recently I acquired an Atik 320e CCD but can't get it to connect. I get an error message. It's an older camera but fully operational it works through Artemis/capture with an older driver from their core legacy download 2.26. (The newer drivers on the web page don't work)
Is it possible to download this older driver onto the pi somehow?
As said I'm a complete novice with this any advice help would be most appreciated.
Kind regards
Paul

rkaczorek, 2021-12-14:
Enable logging on CCD driver and upload log file here. We are using libraries from the ATIK SDK so it should work unless camera is not supported anymore.

Paul47910106, 2021-12-14:
Hi RadekI had just read the instructions on how to generate logs however I am unable to get that far. When connecting to ekos the mount and filter wheel work fine, however I get the following message in relation to the ccdUnable to establish + AtikPlease ensure the device is connected and powered on.Please excuse my ignorance but I'm a complete noob to this. Have been getting great results with a canon dslr and got the opportunity of this dedicated Atik camera. I have read previous threads on the problem but lack the knowledge to understand the solutions.RegardsPaulSent from Samsung Mobile on O2

rkaczorek, 2021-12-15:
Check [this](https://indilib.org/forum/ccds-dslrs/8999-indi-atik-driver-crashed.html) threat on INDI forum.