rkaczorek/astroberry-server #173: ZWO ASI2600MM Pro Issues with 3.5.6

MexicoSkies, 2022-01-20:
After upgrading to the 3.5.6 from 3.5.3, I've experienced an issue with the ASI2600MM Pro camera. A short time into a session, about 15 or 20 minutes, the camera will return images that are obviously compromised somewhere is the processing chain. While this issue is happening, I can carry my notebook computer to the telescope and camera, disconnect the USB3 cable from the camera and plug in another cable to the camera and use the ASI software to capture images that are not compromised. I am also running an ASI290MM Mini for the guide scope with no issues.

I have swapped USB3 cables several times hoping this is the culprit but, still the same issues. I run a powered USB hub as well and I have also tried to put the ASI2600 MM Pro cable directly into the RPi with similar results as well.

Thank you for all of your efforts. This software is incredible!

[image]
[image]
[image]
[image]
.

rkaczorek, 2022-01-22:
Check your power supply. Investigate the logs by running `tail -f /var/log/syslog`
If you get any "Undervoltage detected" your power supply is not providing enough juice to Raspberry.
I don't think it is a software issue. I use this camera myself and have no issue with it at all.

MexicoSkies, 2022-01-22:
Well, of course I wiped the card to try a new install! LOL. Next opportunity for setup, I will let you know if there are any issues with what is showing in the log. I did run the connection through my digital tester and I showed fine. 

Still checking.

[image]

MexicoSkies, 2022-01-23:
It was fine on the V & A but, the connection was very loose. Peculiar because it was an Anderson Connector. When I tried to pull the connector out of the distribution block, the wires just fell out and were obviously cut at the end of the insulation cover. Maybe a bit too aggressive when crimping? I put a couple of new connectors on and everything appears fine, at least as much as I can test in the daylight. Thanks for the keen insight on the power, Radek!

closed 2022-01-23 by rkaczorek

MexicoSkies, 2022-02-03:
Hello Radek,

I thought the power issue fixed everything but, I’ve been having crazy troubles with everything for some reason. I ran the log for the session tonight as well as there is a log for a couple days ago too.

Tonight was crazy guiding and then KStars/Ekos would shut down. Happened 2 or 3 times tonight.

Let me know if I need to post elsewhere.

Thank you,

Gary

rkaczorek, 2022-02-03:
Make sure to install latest updates to KStars and drivers. You need to inspect hardware, including your microSD card.
I have not got any other reports like this recently. You can upload logs here

MexicoSkies, 2022-02-03:
Thanks Radek. I will start that process now.

MexicoSkies, 2022-02-04:
Good Morning Radek,

I just wanted to update you on the shenanigans. I’m sure you see all types of issues and problems but, this is really the first time I have experienced any problem with anything Astroberry in almost 2 years.

When 3.5.7 came out, I heard some good things and I decided to upgrade. At the same time, I wanted to consolidate my accessories on the scope - RPi, 12V power distribution and powered USB hub. I removed the powered hub and rearranged the cabling to fit into the USB ports on the Astroberry. That is when I started to experience the total malfunction for the astrophotography process. Not connecting, crashing, distorted images etc. 

I took your advice and checked all the physical items in the chain. I decided to go back to the original configuration with the powered hub and all devices connected into that hub. I did have a bad electrical connection that was repaired as well.

Last night I was able to image the entire night with no issues. For some reason, the powered USB hub (I think) is integral for my imaging with ASI2600MM Pro, ASI290MM Mini, EQ6-R, EFW & EAF. The interface was the KStars app on iMac and the Astroberry was connected via ethernet cable (for speed). I did have a crash initially when using the Astroberry HotSpot and that’s when I switched to the app. The app is also better for dual monitors.

I hope this will be of some small help when someone has a similar issue in the future.

Thanks again for everything you do and all of your help.

Gary

rkaczorek, 2022-02-06:
Thanks for the update Gary. I hope it will be useful for other users too.
Clear skies!
Radek