rkaczorek/astroberry-server #174: new indi version 1.9.4

luzik, 2022-01-21:
Please bump indi to 1.9.4

rkaczorek, 2022-01-22:
I will for sure. Be patient

luzik, 2022-01-26:
Can't wait :)
There is big celestron-aux rewrite

luzik, 2022-01-28:
I can't find information where are source files for repo packages that jumps into

https://www.astroberry.io/repo buster InRelease
I wanted add self hosted repo with updated packages, but don't want do it from scratch.

rkaczorek, 2022-01-29:
Can you find source files for official repo packages for INDI and INDI 3rd party?

luzik, 2022-01-29:
I can’t find astroberry build scripts for https://www.astroberry.io/repo

Hmm.. is astroberry OpenSource project ? :-)
All open source distributions shares their build scripts like spec files for rpm based distros

rkaczorek, 2022-01-29:
If you want to learn how to build and package software for debian, feel free to study https://wiki.debian.org/Packaging
Regarding your note on opensource, could you share with all of us (so we can do better job) a build script for Raspbian OS?
Also, I would appreciate if you indicate which source files for software distributed as part of Astroberry are missing.

Last but not least, have you noticed updates released today? Does it fix your issue? If so please close the issue.

luzik, 2022-01-29:
Thank you for upgrading! Excuse me for my lack of packaging knowledge. I do have experiences with rpm packaging, and there is a need for spec files. Using link you provided I found something like

https://salsa.debian.org/qt-kde-team/3rdparty/indi/-/tree/master/debian

and after more digging

http://raspbian.raspberrypi.org/raspbian/pool/main/i/indi/

that looks like source files for packaging that I was looking for and didn't find for https://www.astroberry.io/repo

closed 2022-01-29 by luzik

rkaczorek, 2022-01-29:
Just to be clear, as I think we have a misunderstanding here.

- Debian packages are built using files located in debian directory, located in source tree
- Every single package delivered as part of Astroberry provides debian directory in a source tree (debian packaging files). Take a look at the list of custom packages [here](https://github.com/rkaczorek/astroberry-server#where-is-the-source-code-of-the-system).
- All upstream software provides its own debian directory and it is used as is while packaging for Astroberry.
- Astroberry repo does not provide src packages, however every single package is linked to sources maintained on github. You can find neccessary information in a copyright file distributed as part of a package
- Build scripts for entire distribution is not provided along, as they do not exist because... it is manual process

I hope it answers your question

luzik, 2022-01-29:
Yes, thx! I think I confused you using 'build scripts' instead of 'packaging scripts'