rkaczorek/astroberry-server #177: Astroberry Hotspot lost after Screen resolution change in RasPi Config

Agwising, 2022-02-15:
So i tried to change my screen resolution after connecting the astroberry to my iPad Pro. Turns out the screen is not fit for the iPad so i had it changed to the default 960x540 or something (forgot). Then it requested to reboot after config change. So after rebooting, the astroberry hotspot seems to be lost. So i connected the astroberry via HDMI to my tv. After checking the connection, my astroberry seems to have connected miraculously to my home wifi (which I haven’t done successfully previously) so it turns out that the wifi of my RasPi 3B+ cannot have a simultaneous connection of astroberry hotspot with internet connection to my home network.

is it a bug that the network connections were changed after screen resolution config? Thank you!

rkaczorek, 2022-02-15:
It just used existing wifi configuration and connected after reboot. 
Hotspot is disabled by default if any connection to other wifi is detected. It will be enabled automatically if not such connection is detected.
Regarding resolution - it seems easier to keep original resolution (1920x1080) and use client autoscale available only for in-browser connection. If you use VNC autoscale depends on the client you use.

Agwising, 2022-02-23:
Hi! I would like to seek help. I accidentally deleted "Astroberry Hotspot" not it is not appearing. Is there a fix around this withouth flashing a new OS? I can't go through the process of installing the astrometry files again which took me almost the whole night. Thank you!

Agwising, 2022-02-23:
Nevermind! Already solved! In the configuration i just changed from client to hotspot.ü i really wish i could find a tutorial to access the astroberry via Ethernet. I noticed that when my PC is at a distance even less than 5 meters, it is very laggy.

rkaczorek, 2022-02-23:
Accessing Astroberry via ethernet requires just pluging in network cable connected to your router. Nothing more

Agwising, 2022-02-24:
> Accessing Astroberry via ethernet requires just pluging in network cable connected to your router. Nothing more

I mean controlling the astroberry in my PC via ethernet from PC to RPi. I don't know what address to put. 10.42.0.1 won't work.

rkaczorek, 2022-02-24:
Direct connection will not work probably (same as between any two PCs connected this way). You need to configure a DHCP server either on PC or RPi, whcich will automatically assign IP addresses to the devices.
I haven't tested it but you can try connecting PC and RPi directly and check if PC got an and address 169.254.x.x. If so, RPi should have similar address too and the last number will be different only. This would be just w workaround and the proper solution is to connect both devices to a router, which provides DHCP and IP address assignment.