rkaczorek/astroberry-server #179: Ekos not saving settings

mmichal12, 2022-02-23:
After installing latest version of Astroberry my settings in Ekos (mount connection, IP address etc.) are not saved. This hasn't happened with previous version. It is fresh installation with all updates installed.

rkaczorek, 2022-02-24:
Do you mean Ekos settings as seen in KStars or drivers settings as seen in INDI Panel?

mmichal12, 2022-02-24:
INDI panel, sorry for confusion. But not only for one driver, in my case for the mount, NMEA GPS, not sure about DSLR now

rkaczorek, 2022-02-24:
There is no autosave for drivers and never was ;-) You need to Save your parameters on each driver Options tab.

mmichal12, 2022-02-24:
Yes, I tried, setting are not saved, it is always in default config. I tried to purge the config too, from logs I don' see anything helpful to me

rkaczorek, 2022-02-24:
You might inspect or even remove all configuration files from /home/astroberry/.indi (note . in front of indi)
Before doing anything, check permisions to the config files by running `ls -al /home/astroberry/.indi`
If you decide to remove all configs run `rm -rf /home/astroberry/.indi`

mmichal12, 2022-02-24:
ok, thanks, I did some checks and it looks the config file gets saved but doesn't get loaded automatically, when I click Load it loads the config but there is always default config when I start Ekos. Is this how it should be?

rkaczorek, 2022-02-24:
You need to set Load device configuration / On connection in Ekos settings

bbillp, 2022-03-01:
I also discovered a similar problem with the latest KSTARS/EKOS version last night.
Open KSTARS, Open EKOS, Select the previously used profile for Celestron CGX, Nikon D5600 Imager, ASI120 guide.
The D5600 Image Options Tab did not save RAW, NEF, NATIVE .  Also the camera setting for FRAME, WIDTH 600 displays 600 even when attempting to set to 6000 but the camera still works OK, it is just the display that is wrong.

Thanks for the wonderful software, the PLATE Solving is fantastic .