rkaczorek/astroberry-server #181: swig/python detected a memory leak of type 'INDI::BaseDevice::Properties *', no destructor found.

GoateePFE, 2022-05-02:
Today I upgraded from Astroberry 2.0.3 to 2.0.4. I could capture darks and lights just fine. When I set bias to 0.0001 seconds (which worked before on 2.0.3) I just get a spinning icon and the biases do not run. The camera is QHY268C.

syslog shows this error over and over:
swig/python detected a memory leak of type 'INDI::BaseDevice::Properties *', no destructor found.

I see this same error when running darks and lights, but they work. Bias gives a spinning progress icon with no frames captured.

See screenshot for more details:
[image]

rkaczorek, 2022-05-02:
I guess you use pyindi-client on Astroberry. Unfortunately pyindi-client has been abandoned by original author and overtaken by indi project recently. AFAIK it is not operational as for now. Follow the development and issues here https://github.com/indilib/pyindi-client

GoateePFE, 2022-05-02:
@rkaczorek I knew you would have the answer quickly. Thank you. I am not smart enough to change how INDI works in Astroberry. I'm a user. Is there another way I should be using Astroberry that would avoid this issue?

rkaczorek, 2022-05-03:
What do you use pyindi-client for? Astroberry does not use it by default to anything.

GoateePFE, 2022-05-03:
Please help me understand. I cannot explain. All I know is that bias frames with exposure less than 1 second do not work. When I looked at syslog, those are the messages that I found. I have not changed anything in how Astroberry works.

GoateePFE, 2022-05-03:
Looks like the QHY268C has exposure time range: 30µs - 3600sec. Thanks for the links. I found two threads describing the problem. I just did an update on Sunday, so maybe I didn't get the latest version or maybe there was a regression.

[](https://indilib.org/forum/ekos/11660-bias-frames-unable-to-take.html#82589)

[](https://www.indilib.org/forum/ekos/11479-bias-frame-not-captured.html
)

MarticusH, 2022-06-02:
@rkaczorek There is a recent thread about this over on [INDILib](https://www.indilib.org/forum/ekos/10822-kstars-memory-leak-and-crash.html) where folks have run into the same problems; happy to help debug further but definitely a problem (this was causing outright crashes for me on smaller-RAM systems eventually).

I don't think it mattered between a ZWO cam or QHY cam, but could revisit to try and help out.