rkaczorek/astroberry-server #183: Oacapture won't load on new Raspberry Pi 4 astroberry install

stevecoh1, 2022-05-13:
Just got astroberry (v.2.0.4) on new Rasp P1 4. Most of the components work, but not OAcapture. The button on the GUI does not load.

The command line is not working either:
```
$ oacapture
libEGL warning: DR12: failed to authenticate
qt5ct: using qt5ct plugin
qt5ct: D-Bus gloabl menu: no
Segmentation fault
```

How can I get this to work?
Does the system log what is going on?
Thanks.

stevecoh1, 2022-05-13:
> Have you run update on your system after flashing SD card?

yes.
did not help

rkaczorek, 2022-06-18:
I need more details to trace it down. Could you run terminal and stary oacapture from it? Just type `oacapture` in terminal emulator. Copy and post the output from the terminal here.