rkaczorek/astroberry-server #187: No planetary observing software exists for me and my Svbony camera in astroberry

stevecoh1, 2022-06-17:
I have found that there is no planetary observing option for me in astroberry currently. And with my equipment (Celestron 4se) planetary observing is what it does best.

The astroberry wiki lists 2 options for planetary observing

- OaCapture, which does not even load on my Raspberry Pi 4. See [Issue 183](https://github.com/rkaczorek/astroberry-server/issues/183). It has not had a release since Dec. 2020 and it's not clear that it would work with my camera even if it did load.

- FireCapture, which does load but does not support my Svbony camera SV-305. On the [FireCapture website](http://www.firecapture.de/), we see that while Svbony is supported for 64-bit Windows and Linux, it is not supported for the still-32bit raspberry pi

So it seems as if we are waiting for one of several things to happen

- A 64-bit version of astroberry (my Raspberry Pi is a 64-bit processor, but the 64-bit OS has only recently been released from beta (2/2022). After that, one might guess that FireCapture would release a 64-bit Pi version that would be compatible with my Svbony.
- A fix for OaCapture to be released

In the meantime, I have been messing around with CCDCiel/Hnsky/Astap, which are clearly not designed with planetary imaging in mind, though they do offer some minimal support for it. 

It has been a hard slog, but I've managed to capture a few images.

I realize that all of this is beyond your control, but can we anticipate a viable planetary imaging option for astroberry and my equipment? Or is there another option you know of that would support my use case?

It would be great to have an INDI-based planetary imaging software.

If there is an issue with developers not having the Svbony equipment to test with, I'd be glad to help test anything they provide.

Thanks.

rkaczorek, 2022-06-18:
Hi Steve, I replayed in #183. Let's discuss oacapture issue there. I'm a bit surprised as I use it personally without issues. I'm sure we can manage it as it is good planetary software, which supports many cams and filter wheels.
As for FireCapture, I will look into it. 64 bit version of the system is compiling as I write this message. Should be available soon, so stay tuned.
Last but not least, Ekos provides live view for supported cams so you can try it waiting for oacapture or firecapture. It saves ser files so could be quite useful. It's far from being perfect but still can do the job.

rkaczorek, 2022-06-19:
@stevecoh1 have you tried installing FireCapture 2.7.10 from the [original website](http://www.firecapture.de/#download)?